openbmc/openbmc #3499: Witherspoon: phosphor-hwmon-readd[1378]: terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'

gkeishin, 2019-03-15:
Build: https://openpower.xyz/job/openbmc-build/2015/

Version: "2.7.0-dev-90-gae29d00"

```
Mar 15 09:47:51 witherspoon phosphor-hwmon-readd[1378]: terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
Mar 15 09:47:51 witherspoon phosphor-hwmon-readd[1378]:   what():  filesystem error: recursive directory iterator cannot open directory: No such file or directory [/ahb/apb/i2c@1e78a000/i2c-bus@380/tmp275@4a]
```
```
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]: Unable to find hwmon directory from the dev path
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]: Usage: phosphor-hwmon-readd [options]
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]: Options:
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]:     --help               print this menu
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]:     --path=<path>        sysfs location to monitor
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]:     --dev-path=<path>    device path to monitor
Mar 15 10:27:04 witherspoon phosphor-hwmon-readd[2522]: ERROR: Path not specified or invalid.

```

gkeishin, 2019-03-15:
I see this on other systems too..  this is pervassive

spinler, 2019-03-15:
The only hwmon change lately was https://github.com/openbmc/phosphor-hwmon/commit/9e997b4da98725401769ef82fe26e11c46a059d9 - "use filesystem instead of experimental".  I wonder if std::filesystem behaves different than the experimental::filesystem.
FYI @pstrinkle

pstrinkle, 2019-03-15:
> The only hwmon change lately was openbmc/phosphor-hwmon@9e997b4

I'll take a look this morning.

spinler, 2019-03-15:
though that directory it's looking at clearly doesn't exist: `/ahb/apb/i2c@1e78a000/i2c-bus@380/tmp275@4a`

spinler, 2019-03-15:
Turns out the difference is in the /= operator.  

From cpppreference.com, only for std::filesystem:
```
path& operator/=(const path& p);
1) If p.is_absolute() || (p.has_root_name() && p.root_name() != root_name()), then replaces the current path with p as if by operator=(p) and finishes.
```
It doesn't say that for experimental::filesystem.

that breaks this line: https://github.com/openbmc/phosphor-hwmon/blob/master/sysfs.cpp#L170

pstrinkle, 2019-03-15:
Thanks for running that down.

spinler, 2019-03-15:
i can put up a fix in a bit

spinler, 2019-03-15:
Resolved by https://gerrit.openbmc-project.xyz/#/c/openbmc/phosphor-hwmon/+/19288/

closed 2019-04-19 by spinler