openbmc/openbmc #3500: Redfish configuration backup and restore function

krtaylor, 2019-03-18:
Design: 
https://gerrit.openbmc-project.xyz/#/c/openbmc/docs/+/18163/

geissonator, 2019-06-17:
This has been queued up for discussion with DMTF but it will not be making the 2.7 release.

krtaylor, 2019-10-25:
Moving to wiki, as per release planning

AlexanderAmelkin, 2020-01-27:
Any progress on this?

AlexanderAmelkin, 2020-07-30:
I dont' think it's ready to be closed.

geissonator, 2021-06-09:
Any luck getting this defined with DMTF @gtmills ?

gtmills, 2021-06-09:
> Any luck getting this defined with DMTF @gtmills ?

No super low on my priority of things to drive in. Not sure if I will ever get there so if someone needs will have to drive in.

AlexanderAmelkin, 2022-04-19:
I'd prefer to keep this open

edtanous, 2022-05-23:
Next steps on this still involve engaging with Redfish to add schema support for backup/restore.  Is anyone actively working on this to keep it open?

AlexanderAmelkin, 2022-05-24:
We are actively working on this feature in general, but hasn't yet gotten to the Redfish schema part.

williamspatrick, 2023-09-25:
Feature request for an abandoned design that hasn't been touched in a few years.  Closing.

closed 2023-09-25 by williamspatrick