openbmc/openbmc #3503: Aspeed host p2a bridge driver

krtaylor, 2019-03-18:
Design:
https://gerrit.openbmc-project.xyz/#/c/openbmc/docs/+/17274/

krtaylor, 2019-07-31:
I assume this didn't make the release since the design has not merged? Please advise.

pstrinkle, 2019-07-31:
That's done and merged in torvalds/linux.  The design isn't merged, there was also push back on whether a linux driver's design should be merged at all.

pstrinkle, 2019-07-31:
https://github.com/openbmc/linux/commit/dc413a90edbe715bebebe859dc072ef73d490d70#diff-4a2b372bfad52fc1128b013fe5ea7d75

mdmillerii, 2019-07-31:
@pstrinkle you indicated you would update the documentation to match what was merged [here](https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/17274/#message-2d04c8d8_e93be2af).

If you think documenting the ioctl is out of scope, fine, but how about documenting the security concerns and application mitigation (eg limiting access in time to when data is expected)?

closed 2020-02-29 by stale[bot]