openbmc/openbmc #3506: obmc-console: FR: conditionally let socket client access ringbuffer backlog

kunyi731, 2019-03-21:
Currently, when a client connects to the socket it by default is registered as a ringbuffer consumer, whose read location starts from log tail.

However in some scenarios it would be useful to let the client start reading from head and access the backlog buffer contents. A legit use case would be a debug console stream that only connects to the console server after the host system crashed.

Would like to understand whether it is possible to implement a toggle such that client could choose the behavior it needs.

artemsen, 2019-03-25:
Perhaps hostlogger project would be useful in your case:
https://github.com/openbmc/phosphor-hostlogger
It has persistent connection to the console socket, gathered logs are flushed to a file according configured policy.

kunyi731, 2019-04-15:
Thanks for the information. I think we ended up pursuing other directions.

closed 2019-04-15 by kunyi731