openbmc/openbmc #3507: Witherspoon: PNOR priority files wiped clean  post factory reset resulting in boot failure

gkeishin, 2019-03-21:
```
@gkeishin seems the pnor priority files got wiped out so the pnor service doesn’t know which 
version is the active one, let me check if they’re supposed to be restored after
factory reset
```
```
root@witherspoon:~# ls -l /var/lib/obmc/openpower-pnor-code-mgmt/
ls: /var/lib/obmc/openpower-pnor-code-mgmt/: No such file or directory
```

```
Adriana Kobylak   [2 minutes ago]
seems we look at the u-boot env vars to re-create the dbus properties but we don’t write
them back to flash, i can manually create the symlinks, we’ll probably need an enhancement
in there
```

gkeishin, 2019-03-21:
```
root@witherspoon:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-dev"
VERSION_ID="2.7.0-dev-135-gb0afcc1"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
BUILD_ID="2.7.0-dev"
OPENBMC_TARGET_MACHINE="witherspoon"
```
```
root@witherspoon:~# uname -a
Linux witherspoon 5.0.3-b94b74e8b52db91fe4e99e0bb481ec8bf2b5b47c #1 Thu Mar 21 13:38:45 UTC 2019 armv6l GNU/Linux
root@witherspoon:~#

```

rfrandse, 2019-03-26:
<https://gerrit.openbmc-project.xyz/c/openbmc/openpower-pnor-code-mgmt/+/19627> activation_ubi: Set priority in constructor
Fixes:  #3507 Witherspoon: PNOR priority files wiped clean  post factory reset resulting in boot failure

closed 2019-03-27 by geissonator