openbmc/openbmc #3509: Issue in getting session info through sessionless interface (or from host-ipmid).

htnakayrus, 2019-03-25:
IPMI RMCP+ sessions are created and maintained by phosphor-ipmi-net daemon, whereas we need to provide details about the same using phosphor-ipmi-host. Hence IPMI RMCP+ session details have to be exposed through D-Bus interface, so that both phosphor-ipmi-host & phosphr-ipmi-net will be in sync.

bradbishop, 2019-03-25:
This patch might be relevant:

https://gerrit.openbmc-project.xyz/19208

htnakayrus, 2019-03-29:
19209: Create & expose the session object in ipmid-net
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-net-ipmid/+/19209
19633:Update the session/SOL mgmt section in ipmi doc
https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/19633
19208:Create IPMI session interface for host and LAN
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-dbus-interfaces/+/19208
19284:Implementing the session commands for host
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-host-ipmid/+/19284

Fix of this issue. under review.

krtaylor, 2019-07-31:
Most of these patchsets are merged. Is this fixed? Can it be closed? Please advise.

htnakayrus, 2019-09-30:
Please close this. As most of design and code is merged.

closed 2020-03-31 by htnakayrus