openbmc/openbmc #3512: phosphor-inventory-manager: mapper "GetObject" method occasionally fails on xyz.openbmc_project.Inventory.Manager

kunyi731, 2019-03-27:
System: openbmc 2.6
Symptom: occasionally upon BMC reboot, obmc-read-eeprom will fail to get the result from a mapper DBus call:

Mar 07 21:36:04  systemd[1]: Starting Read system/chassis/motherboard EEPROM...
Mar 07 21:36:06  phosphor-read-eeprom[2598]: FRU Data
Mar 07 21:36:06  phosphor-read-eeprom[2598]: Successfully verified area.
Mar 07 21:36:06  phosphor-read-eeprom[2598]: Populated FRU areas
Mar 07 21:36:13  phosphor-read-eeprom[2598]: [[0;1;31m[[0;1;39m[[0;1;31mException from sdbus call[[0m
Mar 07 21:36:13  phosphor-read-eeprom[2598]: sd_bus_call: org.freedesktop.DBus.Error.FileNotFound: path or object not found
Mar 07 21:36:13  phosphor-read-eeprom[2598]: [[0;1;31m[[0;1;39m[[0;1;31mError updating inventory.[[0m
Mar 07 21:36:13  systemd[1]: [[0;1;39m[[0;1;31m[[0;1;39mobmc-read-eeprom@system-chassis-motherboard.service: Main process exit
ed, code=exited, status=1/FAILURE[[0m

this is a call to mapper "GetObject sas /xyz/openbmc_project/inventory 1 xyz.openbmc_project.Inventory.Manager". However both the mapper and inventory manager seem to run fine, and mapper-wait-xyz.openbmc_project.Inventory.Manager exits correctly before obmc-read-eeprom is started.

kunyi731, 2019-03-27:
With some snooping I think the issue is due to a race condition in https://github.com/openbmc/phosphor-inventory-manager/blob/master/manager.cpp#L90. The inventory daemon first acquired the well known D-Bus name, which causes mapper-wait-inventoryblahblah services to finish and start dependent processes.

However, the inventory manager doesn't start processing D-Bus calls until manager.run(), which needs to wait for restore() to finish first. Any mapper call during this period would fail per my understanding.

rfrandse, 2019-03-27:
<https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-inventory-manager/+/19704> manager: Move bus name requesting to after restore
Resolves  #3512 phosphor-inventory-manager: mapper "GetObject" method occasionally fails on xyz.openbmc_project.Inventory.Manager

closed 2019-10-26 by stale[bot]