openbmc/openbmc #3519: Hwmon failing to start up

JesusPhish, 2019-04-02:
Hi all,

I've run into an issue similar to the following but have found no resolution
#3435
#3063 
#3074

I'm setting up a new board and have tried to add a LM75 sensor. I've used a DTS file to set up the sensor and looked at [sensor-architecture.md](https://github.com/openbmc/docs/blob/master/sensor-architecture.md)

A quick summary

- DTS file, associated .conf and .bbappend files are set up
- Can manually read the sensor on the command line by going down into /sys/device/
- Can manually kick off phosphor-hwmon-readd.
- I have in the DTS compatible = "ti,lm75"; but I've also tried this as "national:lm75" and I don't believe it made a difference

From what I can gather, Hwmon.slice starts, the device is set up properly, but the xyz.openbmc_project.Hwmon@.service service doesn't start properly or it does but then exits without an error. 

I've included code snippets, any assistance is appreciated and if anything else needs to be provided please ask. 

BMC Version
```
root@evb-ast2500:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.6.0-rc1"
VERSION_ID="2.6.0-rc1-265-gfb721b2d8-dirty"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.6.0-rc1"
BUILD_ID="2.6.0-rc1"
OPENBMC_TARGET_MACHINE="evb-ast2500"
```

Device Tree
```
&i2c7 {
        status = "okay";

        lm75@4b {
                compatible = "ti,lm75";
                reg = <0x4b>;
         };
};
```

Device in FS
```
root@evb-ast2500:~# ls -l /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@300/lm75\@4b/*
-r--r--r--    1 root     root             8 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/compatible
-r--r--r--    1 root     root             5 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/name
-r--r--r--    1 root     root             4 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/reg
root@evb-ast2500:~# cat /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@300/lm75\@4b/*
ti,lm75
lm75
K
```

Hwmon0 points to lm75
```
root@evb-ast2500:~# cat /sys/class/hwmon/hwmon0/of_node/compatible
ti,lm75
root@evb-ast2500:~# cat /sys/class/hwmon/hwmon0/of_node/name
lm75
root@evb-ast2500:~# hexdump /sys/class/hwmon/hwmon0/of_node/reg
0000000 0000 4b00
0000004
```

conf file in /etc/default/obmc
```
root@evb-ast2500:~# cat /etc/default/obmc/hwmon/ahb/apb/bus\@1e78a000/i2c-bus\@300/lm75\@4b.conf
[... 154 lines cut ...]
```
cat meta-evb/meta-evb-aspeed/meta-evb-ast2500/recipes-phosphor/sensors/phosphor-hwmon_%.bbappend 
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

CHIPS = "bus@1e78a000/i2c-bus@300/lm75@4b"
ITEMSFMT = "ahb/apb/{0}.conf"

ITEMS = "${@compose_list(d, 'ITEMSFMT', 'CHIPS')}"

ENVS = "obmc/hwmon/{0}"
SYSTEMD_ENVIRONMENT_FILE_${PN} += "${@compose_list(d, 'ENVS', 'ITEMS')}"
```

I've also checked that LM75 is enabled in the Kernel
```
 │ Symbol: SENSORS_LM75 [=y]                                                                                                    │  
  │ Type  : tristate                                                                                                             │  
  │ Prompt: National Semiconductor LM75 and compatibles                                                                          │  
  │   Location:                                                                                                                  │  
  │     -> Device Drivers                                                                                                        │  
  │ (1)   -> Hardware Monitoring support (HWMON [=y])                                                                            │  
  │   Defined at drivers/hwmon/Kconfig:1022                                                                                      │  
  │   Depends on: HWMON [=y] && I2C [=y]                                                                                         │  
  │   Selects: REGMAP_I2C [=y] 
```

spinler, 2019-04-02:
Looks like you may have been caught in the middle of the 5.0 kernel device path changes.   Your kernel shows paths like:
```
i2c@1e78a000
```

But you put your conf file in:
```
bus@1e78a000
```

So if you just rename that conf file dir to i2c@1e78a000, you should be good, at least until you move up to the 5.0 kernel.

JesusPhish, 2019-04-02:
> Looks like you may have been caught in the middle of the 5.0 kernel device path changes. Your kernel shows paths like:
>
> ```
> i2c@1e78a000
> ```
> But you put your conf file in:
>
> ```
> bus@1e78a000
> ```
> So [...]

That has completely fixed the issue. Now the sensor shows up as expected.

Thank you very much for the quick response and the solution. 

I first just ran a mkdir on the system to create bus@1e78a000 and then copied the contents of i2c@1e78a000 into it, then reset. Now I've gone and changed 

```CHIPS = "bus@1e78a000/i2c-bus@300/lm75@4b"``` 
to 
```CHIPS = "i2c@1e78a000/i2c-bus@300/lm75@4b"```

and 

```
cat meta-evb/meta-evb-aspeed/meta-evb-ast2500/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/bus@1e78a000/i2c-bus@300/lm75@4b.conf
```
to
```
cat meta-evb/meta-evb-aspeed/meta-evb-ast2500/recipes-phosphor/sensors/phosphor-hwmon/obmc/hwmon/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b.conf
```

closed 2019-04-02 by JesusPhish

AloneWarrior11011, 2025-04-09:
> Hi all,
> 
> I've run into an issue similar to the following but have found no resolution [#3435](https://github.com/openbmc/openbmc/issues/3435) [#3063](https://github.com/openbmc/openbmc/issues/3063) [#3074](https://github.com/openbmc/openbmc/issues/3074)
> 
> I'm setting up a new board and have tried to add a LM75 sensor. I've used a DTS file to set up the sensor and looked at [sensor-architecture.md](https://github.com/openbmc/docs/blob/master/sensor-architecture.md)
> 
> A quick summary
> 
>     * DTS file, associated .conf and .bbappend files are set up
> 
>     * Can manually read the sensor on the command line by going down into /sys/device/
> 
>     * Can manually kick off phosphor-hwmon-readd.
> 
>     * I have in the DTS compatible = "ti,lm75"; but I've also tried this as "national:lm75" and I don't believe it made a difference
> 
> 
> From what I can gather, Hwmon.slice starts, the device is set up properly, but the [xyz.openbmc_project.Hwmon@.service](mailto:xyz.openbmc_project.Hwmon@.service) service doesn't start properly or it does but then exits without an error.
> 
> I've included code snippets, any assistance is appreciated and if anything else needs to be provided please ask.
> 
> BMC Version
> 
> ```
> root@evb-ast2500:~# cat /etc/os-release
> ID="openbmc-phosphor"
> NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
> VERSION="2.6.0-rc1"
> VERSION_ID="2.6.0-rc1-265-gfb721b2d8-dirty"
> PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.6.0-rc1"
> BUILD_ID="2.6.0-rc1"
> OPENBMC_TARGET_MACHINE="evb-ast2500"
> ```
> 
> Device Tree
> 
> ```
> &i2c7 {
>         status = "okay";
> 
>         lm75@4b {
>                 compatible = "ti,lm75";
>                 reg = <0x4b>;
>          };
> };
> ```
> 
> Device in FS
> 
> ```
> root@evb-ast2500:~# ls -l /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@300/lm75\@4b/*
> -r--r--r--    1 root     root             8 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/compatible
> -r--r--r--    1 root     root             5 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/name
> -r--r--r--    1 root     root             4 Apr  2 09:36 /sys/firmware/devicetree/base/ahb/apb/i2c@1e78a000/i2c-bus@300/lm75@4b/reg
> root@evb-ast2500:~# cat /sys/firmware/devicetree/base/ahb/apb/i2c\@1e78a000/i2c-bus\@300/lm75\@4b/*
> ti,lm75
> lm75
> K
> ```
> 
> Hwmon0 points to lm75
> 
> ```
> root@evb-ast2500:~# cat /sys/class/hwmon/hwmon0/of_node/compatible
> ti,lm75
> root@evb-ast2500:~# cat /sys/class/hwmon/hwmon0/of_node/name
> lm75
> root@evb-ast2500:~# hexdump /sys/class/hwmon/hwmon0/of_node/reg
> 0000000 0000 4b00
> 0000004
[... 213 lines cut ...]
> Unload module index
> Unloaded link configuration context.
``` 
on my raspberrypi system udevadm logs as : 
```
Reading rules file: /usr/lib/udev/rules.d/81-net-dhcp.rules
Reading rules file: /etc/udev/rules.d/99-com.rules
/etc/udev/rules.d/99-com.rules:2 Unknown group 'i2c', ignoring.
/etc/udev/rules.d/99-com.rules:3 Unknown group 'spi', ignoring.
/etc/udev/rules.d/99-com.rules:4 Unknown group 'gpio', ignoring.
/etc/udev/rules.d/99-com.rules:10 Unknown group 'gpio', ignoring.
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /etc/udev/rules.d/can.rules
Reading rules file: /etc/udev/rules.d/fb.rules
Reading rules file: /etc/udev/rules.d/touchscreen.rules
hwmon0: /usr/lib/udev/rules.d/70-hwmon.rules:1 RUN '/usr/bin/start_hwmon.sh start %p %E{OF_FULLNAME}'
Properties:
  DEVPATH=/devices/virtual/thermal/thermal_zone0/hwmon0
  ACTION=add
  SUBSYSTEM=hwmon
  USEC_INITIALIZED=307964567
Queued commands:
  RUN{program} : /usr/bin/start_hwmon.sh start /devices/virtual/thermal/thermal_zone0/hwmon0 
Unloaded link configuration context.
```

AloneWarrior11011, 2025-04-22:
> Looks like you may have been caught in the middle of the 5.0 kernel device path changes. Your kernel shows paths like:
>
> ```
> i2c@1e78a000
> ```
>
> But you put your conf file in:
>
> ```
> bus@1e78a000
> ```
>
> So [...]

```
root@raspberrypi3:~# systemctl status   xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-tmp75\x404d.service   
* xyz.openbmc_project.Hwmon@-soc-i2cx407e804000-tmp75x404d.service - Phosphor Hwmon Poller
     Loaded: loaded (/usr/lib/systemd/system/xyz.openbmc_project.Hwmon@.service; static)
     Active: inactive (dead)

Apr 22 07:25:19 raspberrypi3 systemd[1]: Phosphor Hwmon Poller was skipped because of an unmet co...conf).
Hint: Some lines were ellipsized, use -l to show in full.

```
but conf file is located under /etc/default/obmc/hwmon/soc/i2c@7e804000/tmp75@4d.conf 
but phosphor-hwmon-readd expects it from 
/etc/default/obmc/hwmon/soc/i2cx407e804000/tmp75x404d.conf 

I tried alot but cannot fix the problem do help me even if the sensor got exposed on dbus : 

even able to introspect dbus object : 

```busctl introspect xyz.openbmc_project.Hwmon-f6ddcb1b36b4d684fe6d19c36af10f59797735dcf90a85cde8badbd4b618c356.Hwmon1 /xyz/openbmc_project/sensors/temperature/tmp75_sensor
NAME                                                  TYPE      SIGNATURE RESULT/VALUE                             FLAGS
org.freedesktop.DBus.Introspectable                   interface -         -                                        -
.Introspect                                           method    -         s                                        -
org.freedesktop.DBus.Peer                             interface -         -                                        -
.GetMachineId                                         method    -         s                                        -
.Ping                                                 method    -         -                                        -
org.freedesktop.DBus.Properties                       interface -         -                                        -
.Get                                                  method    ss        v                                        -
.GetAll                                               method    s         a{sv}                                    -
.Set                                                  method    ssv       -                                        -
.PropertiesChanged                                    signal    sa{sv}as  -                                        -
xyz.openbmc_project.Sensor.Threshold.Critical         interface -         -                                        -
.CriticalAlarmHigh                                    property  b         false                                    emits-change writable
.CriticalAlarmLow                                     property  b         false                                    emits-change writable
.CriticalHigh                                         property  d         40                                       emits-change writable
.CriticalLow                                          property  d         0                                        emits-change writable
.CriticalHighAlarmAsserted                            signal    d         -                                        -
.CriticalHighAlarmDeasserted                          signal    d         -                                        -
.CriticalLowAlarmAsserted                             signal    d         -                                        -
.CriticalLowAlarmDeasserted                           signal    d         -                                        -
xyz.openbmc_project.Sensor.Threshold.Warning          interface -         -                                        -
.WarningAlarmHigh                                     property  b         true                                     emits-change writable
.WarningAlarmLow                                      property  b         false                                    emits-change writable
.WarningHigh                                          property  d         10                                       emits-change writable
.WarningLow                                           property  d         0                                        emits-change writable
.WarningHighAlarmAsserted                             signal    d         -                                        -
.WarningHighAlarmDeasserted                           signal    d         -                                        -
.WarningLowAlarmAsserted                              signal    d         -                                        -
.WarningLowAlarmDeasserted                            signal    d         -                                        -
xyz.openbmc_project.Sensor.Value                      interface -         -                                        -
.MaxValue                                             property  d         inf                                      emits-change writable
.MinValue                                             property  d         -inf                                     emits-change writable
.Unit                                                 property  s         "xyz.openbmc_project.Sensor.Value.Uni... emits-change writable
.Value                                                property  d         26.125                                   emits-change writable
[... cut ...]
    "path": "system",
    "interface": "xyz.openbmc_project.Inventory.Decorator.Asset",
    "property": "Model",
    "values": ["RaspberryPi3"]
  },
  "associations": [
    {
      "path": "system,
      "endpoints": [
        {
          "types": { "fType": "sensors", "rType": "inventory" },
          "paths": ["/xyz/openbmc_project/sensors/temperature/tmp75_sensor"]
        }
      ]
    }
  ]
}
```
the main issue was : 
  xyz.openbmc_project.Hwmon@-soc-i2c\x407e804000-tmp75\x404d.service
service failing to start