openbmc/openbmc #3521: inventory: invalid filenames in inventory persisted path

kunyi731, 2019-04-09:
I saw some invalid files generated from time to time:

# ls -lR /var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/
/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/:
drwxr-xr-x    3 root     root             0 Apr  8 18:19 system

/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system:
drwxr-xr-x    3 root     root             0 Mar 21 03:29 chassis
-rw-r--r--    1 root     root            81 Apr  2 20:16 xyz.open? ???1??xyz.Inve???????lxyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.open5????:*Fxyz.Inve?-(?????xyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Apr  2 20:17 xyz.openA??????yxyz.Invebmc_4???xyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.openL???Y???xyz.Inve&??????Cxyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Apr  8 18:19 xyz.openbmc_project.Inventory.Decorator.AssetTag
-rw-r--r--    1 root     root            81 Apr  8 18:19 xyz.openbmc_projxyz.Inventory.Dexyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Apr  2 20:16 xyz.openbmc_?osDxyz.Inventorproject.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar 21 03:31 xyz.opens?38?? ?xyz.Inve`W?V?'??xyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.open~??? no?xyz.Inve?? %.??yxyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.open??.\?m??xyz.Inve??F??^??xyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.open??,?&???xyz.Inve???+?#??xyz.tor.AssetTag
-rw-r--r--    1 root     root            81 Mar  8 00:38 xyz.open?J#}$l0?xyz.Inve???pQU??xyz.tor.AssetTag

/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system/chassis:
drwxr-xr-x    2 root     root             0 Apr  8 18:19 motherboard

/var/lib/phosphor-inventory-manager/xyz/openbmc_project/inventory/system/chassis/motherboard:
-rw-r--r--    1 root     root           223 Apr  2 20:17 xyz.openF???V??8xyz.Invebmc_projxyz.tor.Asset
-rw-r--r--    1 root     root           223 Apr  8 18:19 xyz.openbmc_project.Inventory.Decorator.Asset
-rw-r--r--    1 root     root           113 Apr  8 18:19 xyz.openbmc_project.Inventory.Item
-rw-r--r--    1 root     root           223 Apr  8 18:19 xyz.openbmc_projxyz.Inventory.Deect.tor.Asset
-rw-r--r--    1 root     root           223 Mar  8 00:38 xyz.openh???????xyz.Inve?`?+?)??xyz.tor.Asset
-rw-r--r--    1 root     root           223 Mar  9 07:39 xyz.open?R2??V?~xyz.Inve?????:Kzxyz.tor.Asset
-rw-r--r--    1 root     root           223 Mar  8 00:38 xyz.open?H ?f???xyz.Inve?Gr???eixyz.tor.Asset
-rw-r--r--    1 root     root           223 Apr  2 20:17 xyz.open??S??<??xyz.Invebmc_???Oxyz.tor.Asset
-rw-r--r--    1 root     root           223 Mar  8 00:38 xyz.open???C?q??xyz.Invezf?,?ao'xyz.tor.Asset

mdmillerii, 2019-04-09:
We saw something similar during early integration of the BMC flash driver when the bmc code had a generic arm `memcpy` routine.  The generic routine will read some bytes more than once when reading from unaligned memory (which `jffs2` will do) and caused the flash to be misread, causing similar file system corruption.

What BMC processor do you have?  What is your flash driver?

mdmillerii, 2019-04-15:
@kunyi731 can you comment?  I suspect your flash `mtd` driver is or was incompatible with `jffs2` format.

kunyi731, 2019-04-15:
> @kunyi731 can you comment? I suspect your flash `mtd` driver is or was incompatible with `jffs2` format.

Thanks for the information. The MTD driver is based on the Nuvoton linux driver in https://github.com/Nuvoton-Israel/linux/blob/d5efc19dc6c35e2a3b426b2c76f33b771c3a428e/drivers/mtd/spi-nor/npcm-fiu.c. 

Do you still have a link to the fix in the aspeed mtd driver by any chance?

mdmillerii, 2019-04-18:
The code was changed before merged upstream and it has been re-based a few times, but here is a comment describing how it is broken:

[aspeed-smc.c line 236-243](https://github.com/openbmc/linux/blob/280a5e83425e0e597a3765037a03cd03e80f3111/drivers/mtd/spi-nor/aspeed-smc.c#L236-L243)

I see the `npcm-fiu` driver is calling [`memcpy_fromio`](https://github.com/openbmc/linux/blob/ed068475e807a70e1c6dfaa4782ed4d252967ed8/drivers/mtd/spi-nor/npcm-fiu.c#L264) and [`memcpy_toio`](https://github.com/openbmc/linux/blob/ed068475e807a70e1c6dfaa4782ed4d252967ed8/drivers/mtd/spi-nor/npcm-fiu.c#L276).

I think other flash drivers have also had similar problems but may not have reached root cause.

mdmillerii, 2019-04-19:
@cyrilbur-ibm points out that routine is sutiable for reading from a fifo but does not increement the source address.

kunyi731, 2019-04-20:
> @cyrilbur-ibm points out that routine is sutiable for reading from a fifo but does not increement the source address.

Thanks for the pointer! Will contact Nuvoton and verify that `memcpy_*` ops are the culprit.

closed 2019-11-19 by stale[bot]