openbmc/openbmc #3522: BMC performance indication via event when CPU / Mem / FS usage goes > 90%

sivassrr, 2019-04-11:


sivassrr, 2019-04-11:
This is based on recent mail exchange in community regarding BMC performance metrics.

From end user perspective,  If BMC performance / usages reaches consistently > 90% of BMC CPU utilization / BMC Memory / BMC file system then we should have way to get an event accordingly. This will help end user. I feel this is higher priority.
May be based on the event, involved application should try to correct itself.

Check for BMC Performance at regular interval

- [ ] BMC CPU % usage  consistently > 90% 

- [ ] BMC Memory % usage consistently > 90% 

or 

- [ ] BMC filesystems which need to be monitor and its  % usage consistently > 90%

Need an event and in fact application should take this as input and check whether they are causing the high performance and if they can cancel or overcome they should try to correct it.

sivassrr, 2019-04-26:
Basically would like to have regular monitor 
```

- [ ] say at-least once an hour to check the performance. 

- [ ] when BMC performance goes >= 90 % issue a warning event

- [ ] when BMC performance goes >=95% issue a medium event 

```

sivassrr, 2019-07-09:
I think this change is needed as early as possible,
Based on discussion with @anoo1 She had mentioned that if file system is full and if we ignore then we may not be able to login as well. ```Escape character is '^]'.

witherspoon login: root
[425280.209033] systemd-journald[821]: Failed to rotate /var/log/journal/9ca565cbdddf4a3094cfd1b47e55f65b/system.journal: Read-only file system
[425280.228731] systemd-journald[821]: Failed to write entry (22 items, 686 bytes), ignoring: Input/output error

Login incorrect
witherspoon login: [425284.011862] systemd-journald[821]: Failed to rotate /var/log/journal/9ca565cbdddf4a3094cfd1b47e55f65b/system.journal: Read-only file system
[425284.028440] systemd-journald[821]: Failed to write entry (22 items, 664 bytes), ignoring: Input/output error```

vishwabmc, 2019-07-09:
@sivassrr this is part of the BMC health and telemetry workgroup.

closed 2020-02-07 by stale[bot]