openbmc/openbmc #3525: libdnf: Requires swig 3.0 but doesn't specify

wak-google, 2019-04-11:
Our system swig2.0 satisfies the dependency during the configure stage, but the build fails because it depends on features in swig 3

joepow2, 2019-06-12:
Is this reason leading to build image failed?

`| ERROR: Function failed: do_compile (log file is located at /openbmc/build/tmp/work/x86_64-linux/libdnf-native/0.28.1-r0/temp/log.do_compile.89530)
ERROR: Task (virtual:native:/openbmc/meta/recipes-devtools/libdnf/libdnf_0.28.1.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2794 tasks of which 2780 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/openbmc/meta/recipes-devtools/libdnf/libdnf_0.28.1.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.`

mdmillerii, 2019-06-12:
@joepow2 you should look at the errors in the log file to diagnosis the failure..  
The log for the failing step is mentioned above:

> `| ERROR: Function failed: do_compile (log file is located at /openbmc/build/tmp/work/x86_64-linux/libdnf-native/0.28.1-r0/temp/log.do_compile.89530)`

closed 2020-01-11 by stale[bot]