openbmc/openbmc #3526: Witherspoon: OCC1 disabled post BMC reset when Host is booted

gkeishin, 2019-04-12:
Test:
> Boot Host to OS
> Reboot BMC

Expects: Host OS to be up and running and OCC be active post reset.

```
    "/org/open_power/control/occ0": {
      "OccActive": true,
      "ThrottleMemTemp": false,
      "ThrottleProcPower": true,
      "ThrottleProcTemp": true
    },
    "/org/open_power/control/occ1": {
      "OccActive": false,
      "ThrottleMemTemp": false,
      "ThrottleProcPower": false,
      "ThrottleProcTemp": false
    },
```
```
root@witherspoon:~# journalctl --no-pager -b | grep -i occ
Apr 12 18:40:17 witherspoon kernel: occ-hwmon occ-hwmon.1: OCC found, code level: op_920_180225a
Apr 12 18:42:03 witherspoon systemd[1]: Starting OpenPOWER OCC controller...
Apr 12 18:42:10 witherspoon systemd[1]: Started OpenPOWER OCC controller.
Apr 12 18:43:06 witherspoon kernel: occ-hwmon occ-hwmon.1: OCC found, code level: op_920_180225a
Apr 12 18:43:07 witherspoon kernel: occ-hwmon occ-hwmon.2: OCC found, code level: op_920_180225a
Apr 12 18:43:08 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@-devices-platform-gpio\x2dfsi-fsi0-slave@00\x2d\x2d00-00\x2d\x2d00\x2d\x2d00\x2d\x2d06-sbefifo1\x2ddev0-occ\x2dhwmon.1.service: Main process exited, code=killed, status=15/TERM
Apr 12 18:43:08 witherspoon systemd[1]: xyz.openbmc_project.Hwmon@-devices-platform-gpio\x2dfsi-fsi0-slave@00\x2d\x2d00-00\x2d\x2d00\x2d\x2d00\x2d\x2d06-sbefifo1\x2ddev0-occ\x2dhwmon.1.service: Succeeded.
root@witherspoon:~#
```

mine260309, 2019-04-15:
```
      "ThrottleProcPower": true,
      "ThrottleProcTemp": true
```
About this, it's a bug in occ driver, and I have a fix to be sent to upstream.

About why `OccActive` is false for occ1, it may be related to the host state check:
```
root@romulus:/tmp# obmcutil state
CurrentBMCState     : xyz.openbmc_project.State.BMC.BMCState.Ready
CurrentPowerState   : xyz.openbmc_project.State.Chassis.PowerState.On
CurrentHostState    : xyz.openbmc_project.State.Host.HostState.Off
```
The host is actually running, but host state manager thought it's off.

geissonator, 2019-04-15:
Do you have the full journal @gkeishin? We've had issues in this path with sdbus/mapper timings (especially when occ is hitting issues) causing the host state to be off.

gkeishin, 2019-04-15:
- [ ] Journald 
[OCC1_not_active.txt](https://github.com/openbmc/openbmc/files/3080382/OCC1_not_active.txt)

geissonator, 2019-04-15:
Ahh, we've got an internal defect for this one:

```
Apr 12 18:42:31 witherspoon phosphor-host-check[1653]: Check if host is running
...
Apr 12 18:42:49 witherspoon ipmid[1568]: Command in process, no attention
Apr 12 18:42:49 witherspoon systemd[1]: Started Wait for /org/openbmc/mboxd.
Apr 12 18:42:49 witherspoon systemd[1]: Starting Reload mboxd during power on...
Apr 12 18:42:52 witherspoon systemd-networkd[1749]: eth0: Configured
Apr 12 18:42:52 witherspoon mboxctl[1753]: Reset: Success
Apr 12 18:42:52 witherspoon systemd[1]: Started Reload mboxd during power on.
Apr 12 18:42:54 witherspoon phosphor-network-manager[1479]: Refreshing the objects.
Apr 12 18:42:55 witherspoon phosphor-network-manager[1479]: Refreshing complete.
Apr 12 18:43:01 witherspoon ipmid[1568]: Host control timeout hit!
Apr 12 18:43:01 witherspoon phosphor-host-check[1653]: Host is not running!
Apr 12 18:43:01 witherspoon ipmid[1568]: Failed to deliver host command
```

geissonator, 2019-04-22:
I did some more digging into this. It's something within the path where the BMC uses the SMS attention to trigger a host query.

In this path, the BMC triggers the SMS, the host then comes down and queries the reason for the SMS and acts accordingly.  You can test this path by running "phosphor-host-check" when the system is up and running. Below is the debug trace from btbridged during the fail.

```
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.670529124] Sending SMS_ATN ioctl (45312) to /dev/ipmi-bt-host
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.670969568] Processed 1 dbus events
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.672195102] Sending dbus signal with seq 0x59, netfn 0x06, lun 0x00, cmd 0x31
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.677291219] Received a dbus response for msg with seq 0x59
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.677518047] Processed 1 dbus events
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.677705885] Disabling timer, no elements remain in queue
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.677897885] Successfully wrote 6 of 6 bytes to /dev/ipmi-bt-host
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.678262571] Turning off POLLOUT for the BT in poll()
Apr 19 20:29:46 witherspoon btbridged[1620]: [BTBRIDGED 1555705786.678419218] Completed request with seq 0x59, netfn 0x07, lun 0x00, cmd 0x31, cc 0x00
Apr 19 20:30:16 witherspoon ipmid[1624]: Host control timeout hit!
```

The data seems to indicate the SMS attention was received by the host and the host came down and read the data (0x06:0x31    //<App>:<Get Message Flags>). My guess is the recent refactor to ipmid, possibly the host read not getting routed to the correct place within ipmid?

geissonator, 2019-04-22:
Oh, pretty sure this change is the issue: https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-host-ipmid/+/16606/12/systemintfcmds.cpp#90

We used to hard code a response to the host saying they needed to come down and read more data. Looks like that was changed to return 0 so the host will not come down and process our data. @vmauery can you give some background? I see there was pretty vague TODO in there so maybe you were just trying to clean that up. We count on that being set though for the host communication process here.

rfrandse, 2019-04-22:
<https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-host-ipmid/+/20816> Ensure host is told data available on SMS_ATN
Resolves  #3526 Witherspoon: host in incorrect state after bmc reboot with host up

closed 2019-04-23 by geissonator