openbmc/openbmc #3527: How to decide which NIC to set ntpservers on?

xuweibj, 2019-04-17:
The scenario is like this:
Have 2 NICs on openbmc
1. eth0: IP is 169.254.206.172
2. eth0_1: IP is 172.16.3.36 -- set

Now want to set ntpservers for openbmc, which NIC we should set on?
It seems that we set ntpservers on eth0_1, which we set our IP on, but it does not work. But set on eth0, it works.
@gurevichmark  @whowutwut @zet809

gurevichmark, 2019-04-17:
@xuweibj Do you know how to verify that ntpservers are working ?

gurevichmark, 2019-04-17:
I connected to one of Mark P machines to look at the web interface and NTP Server panels:

[image]

Clicking "Add NTP Server" button:

[image]

xuweibj, 2019-04-18:
@gurevichmark , I don't know, so hope could get exact message here.

Obihoernchen, 2019-04-18:
Set some wrong time manually and switch back to NTP, then check if time changes after node reboot.

gurevichmark, 2019-04-18:
Used BMC Web Interface for `mid05tor12cn05` which has `eth0` and `eth0_11` to set NTP servers attribute.
The Web interface sets it to `eth0`, but our `rspconfig ntpservers` command sets it to `eth0_11`.

closed 2019-11-17 by stale[bot]