openbmc/openbmc #3529: Witherspoon: The system shut down due to not enough functional fans

gkeishin, 2019-04-19:
Boot to host OS and then reset BMC.  We are in bit mess once we reset..

@spinler @msbarth @geissonator we discussed on it earlier. So its time to look at the root cause for this problem.

```
Apr 19 17:25:46 witherspoon phosphor-dbus-monitor[1393]: Request shutdown of system. There are not enough functional fans.
Apr 19 17:25:46 witherspoon phosphor-dbus-monitor[1393]: The system shut down due to not enough functional fans

```

```
Apr 19 17:25:29 witherspoon phosphor-dbus-monitor[1393]: The inventory is nonfunctional.
Apr 19 17:25:29 witherspoon phosphor-fru-fault-monitor[1545]: /xyz/openbmc_project/logging/entry/3 created
Apr 19 17:25:29 witherspoon phosphor-fru-fault-monitor[1545]: /xyz/openbmc_project/logging/entry/4 created
Apr 19 17:25:29 witherspoon phosphor-fru-fault-monitor[1545]: /xyz/openbmc_project/logging/entry/5 created
```

spinler, 2019-04-19:
Resolved by https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-inventory-manager/+/20764

closed 2019-04-19 by spinler

gkeishin, 2019-04-19:
Sweet.. Thanks @spinler