openbmc/openbmc #3531: Failure to boot on Talos II with latest OpenBMC and op-build (HBBL failure)

madscientist159, 2019-04-20:
It appears that the latest OpenBMC and op-build releases do not work together on Talos II / Romulus type systems, specifically in an environment where the SBE on the processors dates from June 2018.  I am also unable to get a current op-build image to boot using the old known-good OpenBMC build from June 2018, however I suspect this is a known problem due to the ASpeed CVE from late 2018.

Thus far I have isolated the following symptoms:
* SBE is kicked off by the BMC, it reaches ISTEP 5,2 and transitions to SBE_STATE_RUNTIME successfully.
* Hostboot never shows up on the console -- the machine hangs at this point.  CFAM 0x283c shows (somewhat iffy) data indicating it's stuck in non-existent ISTEP 6,0.
* mboxd trace / debug mode shows zero traffic originating from the host.  mboxd is successfully transitioned into the Flash mapping state before the IPL failure and does not exit this state.

I don't quite understand how we can reach / complete ISTEP 5,2 without generating LPC mboxd traffic of any type.  Is there a known problem with the older SBE attempting to access the CVE-protected BMC?

madscientist159, 2019-04-20:
@stewart-ibm @sgupta2m any known issues with this setup I should be aware of?

mine260309, 2019-04-22:
The LPC mbox is replaced by IPMI mbox.
Please check if your BMC build picks the [hiomap](https://github.com/openbmc/openpower-host-ipmi-flash)

rajadas2, 2019-04-22:
It can be an issue with the old SBE SEEPROM (which has the HBBL image) trying to boot up with a new PNOR image build from op-build ? is that the issue ?
I would suggest you to update the SBE Seeprom with help of cronus to the version contained in PNOR (sbe seeprom) and see how it goes.

amboar, 2019-04-22:
I can look into this, but I'm on leave until the 30th

madscientist159, 2019-04-22:
The latest I have on this issue is that it was triggered by the GCC8 patches at https://github.com/open-power/op-build/pull/2803 .  However, after reverting to GCC6 and fixing up a few other problems we're still blocked on https://github.com/open-power/hostboot/issues/173

dcrowell77, 2019-04-22:
Another hint of where the host is sitting:
  getscom pu 41010A89
That will show an ascii string to show if you are in "hostboot" or "bootload".

I agree that the istep 6.0 value in 283C seems dubious, what is the complete value in the register?  If the first byte is not 0xAA then it is not istep data, it is boot configuration information that is passed in the same register.  See https://github.com/open-power/hostboot/blob/master/src/include/usr/initservice/mboxRegs.H

Why are you pulling in an unmerged set of changes (the GCC8 stuff)?

madscientist159, 2019-04-22:
> Why are you pulling in an unmerged set of changes (the GCC8 stuff)?

Hostboot doesn't build on GCC7 -- see https://github.com/open-power/hostboot/issues/172 .  Reverting to GCC6 was a last resort.

closed 2019-12-01 by stale[bot]