openbmc/openbmc #3532: witherspoon ipmid: Failed to get bus name

gkeishin, 2019-04-22:
```
Apr 22 17:12:49 witherspoon ipmid[1463]: Failed to get bus name
Apr 22 17:12:49 witherspoon ipmid[1463]: Unsupported parameter
```
Seen during booting...

```
{
    "_PID" : "1588",
    "CODE_LINE" : "105",
    "ERROR" : "No such file or directory",
    "PRIORITY" : "3",
    "_HOSTNAME" : "witherspoon",
    "SYSLOG_IDENTIFIER" : "netipmid",
    "CODE_FUNC" : "helper_log",
    "_UID" : "0",
    "MESSAGE" : "Failed to get bus name",
    "_CAP_EFFECTIVE" : "3fffffffff",
    "__CURSOR" : "s=0eae46b8909c496c8c5132c8b6bcc646;i=1b58;b=be231487c73d4dbda433c001c54a0ee2;m=abfb77c;t=587217373c224;x=c1484e17df59ed25",
    "_SYSTEMD_CGROUP" : "/system.slice/phosphor-ipmi-net.service",
    "_MACHINE_ID" : "789e9af2788f4ac7b0ad2441e61b85d9",
    "_GID" : "0",
    "_SYSTEMD_SLICE" : "system.slice",
    "_BOOT_ID" : "be231487c73d4dbda433c001c54a0ee2",
    "CODE_FILE" : "/usr/src/debug/phosphor-ipmi-net/1.0+gitAUTOINC+8af90ebcc5-r1/recipe-sysroot/usr/include/phosphor-logging/log.hpp",
    "PATH" : "/org/openbmc/control/chassis0",
    "_COMM" : "netipmid",
    "__REALTIME_TIMESTAMP" : "1555952624189988",
    "_EXE" : "/usr/bin/netipmid",
    "_CMDLINE" : "netipmid",
    "TRANSACTION_ID" : "544788054",
    "_SYSTEMD_INVOCATION_ID" : "809bceb55e2e4d97a9f2a2fbc9bd54eb",
    "_SOURCE_REALTIME_TIMESTAMP" : "1555952624189887",
    "__MONOTONIC_TIMESTAMP" : "180336508",
    "_SYSTEMD_UNIT" : "phosphor-ipmi-net.service",
    "_TRANSPORT" : "journal"
}
```

looks like something to do with UUID.

geissonator, 2019-04-22:
We removed the legacy python app that hosts the UUID object. We need to come up with a design for UUID and get it implemented. IPMI seems to just use a default when it can't find one so we should be ok until we figure it out.  FYI @tomjoseph83 @vmauery @nasamuffin

vmauery, 2019-04-22:
The netipmid guid code needs to be re-worked to make sure it matches the ipmid code.

For the BMC's UUID, I think that it should be keyed off the systemd machine-id via sd_id128_get_machine_app_specific https://www.freedesktop.org/software/systemd/man/sd_id128_get_machine_app_specific.html

A quick call to the settings daemon could tell if it was set. If not present, the ipmid could call sd_id128* and create the value and stuff it into the setting. Then all other interested parties could find it.

That is just my opinion. I have not discussed it with anyone else. :)

tomjoseph83, 2019-04-23:
Using machine-id looks fine after covering the man page. I assume the settings daemon will kick-in before the netipmid/ipmid comes up.

geissonator, 2019-10-23:
Still seeing this with 2.8.0-dev-675

closed 2020-05-25 by stale[bot]