openbmc/openbmc #3533: nscd: failure while reading configuration file; this is fat

gkeishin, 2019-04-23:
Persistent failure on boot to standby in master
```
Apr 23 05:38:26 openbmc nscd[1159]: /usr/sbin/nscd: Unknown option: 9a712ee08616432cadbdcc128d7bd81c
Apr 23 05:38:26 openbmc nscd[1159]: /usr/sbin/nscd: failure while reading configuration file; this is fat
```
```
root@openbmc:~# systemctl status nscd.service | cat
● nscd.service - Name Service Cache Daemon
   Loaded: loaded (/lib/systemd/system/nscd.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2019-04-23 05:39:08 UTC; 2min 2s ago
  Process: 1300 ExecStart=/usr/sbin/nscd (code=exited, status=1/FAILURE)

Apr 23 05:39:08 openbmc systemd[1]: nscd.service: Service RestartSec=1s expired, scheduling restart.
Apr 23 05:39:08 openbmc systemd[1]: nscd.service: Scheduled restart job, restart counter is at 4.
Apr 23 05:39:08 openbmc systemd[1]: Stopped Name Service Cache Daemon.
Apr 23 05:39:08 openbmc systemd[1]: nscd.service: Start request repeated too quickly.
Apr 23 05:39:08 openbmc systemd[1]: nscd.service: Failed with result 'exit-code'.
Apr 23 05:39:08 openbmc systemd[1]: Failed to start Name Service Cache Daemon.
root@openbmc:~#
```

ratagupt, 2019-04-23:
@gkeishin 
Some how nscd.conf is corrupted, it is not having the correct data.
root@openbmc:~# cat /etc/nscd.conf 
9a712ee08616432cadbdcc128d7bd81c
root@openbmc:~# 

In any good system it should have data like below.
```
	enable-cache		passwd		yes
	positive-time-to-live	passwd		600
	negative-time-to-live	passwd		20
	suggested-size		passwd		211
	check-files		passwd		yes
	persistent		passwd		yes
	shared			passwd		yes
	max-db-size		passwd		33554432
	auto-propagate		passwd		yes
```

gkeishin, 2019-04-23:
Factory reset it clean and it started seeing it clean and the nscd.conf is no longer junk.. wondering if there are due to code update for different images..

ratagupt, 2019-04-23:
hmm, nscd file is not getting touched by any application on the BMC, why after code update is getting changed? 
@anoo can you please take a look on this?

anoo1, 2019-04-23:
Was unable to recreate so seems it may be intermittent. Let's keep an eye for this issue. Also would be good to check if the nscd.conf file exists in /var/persist/etc/ at the time of failure which would tell that an application or user modified the file.

gkeishin, 2019-07-09:
@anoo1 looks like we had this since.. so will close it up once we find the root cause for it.

gkeishin, 2019-07-09:
Closing this one since I see we are tracking here https://github.com/openbmc/openbmc/issues/3578

closed 2019-07-09 by gkeishin