openbmc/openbmc #3536: WebUI - Sensor Info Not showing

akashgj, 2019-04-29:
Web user interface is unable to load the sensor readings. Through ipmitool commands sensor readings are available.

gtmills, 2019-04-29:
@akashgj What system? 
What build?

akashgj, 2019-04-30:
root@zaius:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="0.1.0"
VERSION_ID="0.1.0"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 0.1.0"
BUILD_ID="20190424124810"
OPENBMC_TARGET_MACHINE="zaius"

JesusPhish, 2019-05-02:
I ran into this issue as well, testing on both Chrome and FF. The page loads as far as displaying everything but the populated sensor table, at which point it just spins forever. 

I tried with the latest [commit](https://github.com/openbmc/phosphor-webui/commit/615a2f896951643ae32e35b308e9bff649caf331) to Phosphor-webui but found that the issue persists, so I used an older commit from a different checkout which I had known to work. (75494d4a69260f8e192d250d1f8c7813e4c759e1)

Changing back to that revision got the sensors to display via the webui again. 

I'm not sure at which point in between these two commits the issue first developed though but it looks like the latest commit to revert the previous changes aren't enough to get the sensors to pop back up.

gtmills, 2019-05-03:
We revert moving the sensor page to Redfish for a separate reason. 
https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-webui/+/20756
Let me know if that fixed the problem.

gtmills, 2019-05-03:
Do `curl -k https://${bmc}/redfish/v1/Chassis/{ChassisId}/Thermal` or `curl -k https://${bmc}/redfish/v1/Chassis/{ChassisId}/Power` return an error? 
Eventually, the GUI sensor page will look at: 
`/redfish/v1/Chassis/{ChassisId}/Thermal`
`/redfish/v1/Chassis/{ChassisId}/Power`
`/redfish/v1/Chassis/{ChassisId}/Sensors`

JesusPhish, 2019-05-07:
> We revert moving the sensor page to Redfish for a separate reason.
> https://gerrit.openbmc-project.xyz/c/openbmc/phosphor-webui/+/20756
> That fix the problem?

After cleaning the build directory out and re-updating to the latest commit quoted here, the sensor info page appears to be working for me at least.

closed 2019-12-06 by stale[bot]