openbmc/openbmc #3540: "Curl ... -X PUT...."  response  "Unauthorized" error

deanyin123, 2019-05-07:
openbmc version: 2.7.0-dev-347-g6d92949-dirty

```
curl -c cjar -b cjar -k -X POST -H "Content-Type:application/json" -d '{"data":["root","0penBmc"]}' https://192.168.3.77/login                                                             {
  "data": "User 'root' logged in",
  "message": "200 OK",
  "status": "ok"
}
```
```
curl b cjar -k  -H "Content-Type:application/json" -X PUT -d '{"data":"xyz.openbmc_project.State.Host.Transition.On"}' https://192.168.3.77/xyz/openbmc_project/state/host0/attr/RequestedHostTransition
Unauthorized
```

mine260309, 2019-05-07:
Please use "Token based authentication" described in https://github.com/openbmc/docs/blob/master/REST-cheatsheet.md#establish-rest-connection-session

deanyin123, 2019-05-08:
> Please use "Token based authentication" described in https://github.com/openbmc/docs/blob/master/REST-cheatsheet.md#establish-rest-connection-session

Thanks, it works now!

closed 2019-05-08 by deanyin123