openbmc/openbmc #3541: not support flash chip for w25m512jv

zhanganha, 2019-05-08:
When using w25m512jv this flash chip. the aspeed smc driver get the wrong size of the chip, its
 size is 64M, but the space identified by the aspeed smc driver is 32M, what can i do to fix it

[image]

shenki, 2019-05-08:
The chip is defined in drivers/mtd/spi-nor/spi-nor.c.

Looking at the version in 5.0, it appears to be correctly defined as 64MB:

```
        { "w25m512jv", INFO(0xef7119, 0, 64 * 1024, 1024,
                        SECT_4K | SPI_NOR_QUAD_READ | SPI_NOR_DUAL_READ) },
```

What is your kernel version?

zhanganha, 2019-05-08:
in 5.0.13

MichaelWang2017, 2019-05-16:
` { "w25m512jv", INFO(0xef7119, 0, 64 * 1024, 1024,
                        SECT_4K | SPI_NOR_QUAD_READ | SPI_NOR_DUAL_READ) },`
Yes, this params is correct, but spi_nor_scan ->spi_nor_init_params -> spi_nor_parse_sfdp -> spi_nor_parse_bfpt will change params->size from 67108864 to 33554432 by reading nor flash, so the code can not identify w25m512jv as 64M. w25m512jv consists of two w25q256jv.
[image]

MichaelWang2017, 2019-05-16:
mx66l51235f can be correctly identified as 64M.
[image]

mdmillerii, 2019-05-16:
Looking briefly at the data sheet, `w25m512jv` is a multiple die package made up of two (stacked) `W25Q256JV` die   with each die being 32MB, and uses a new command to switch between the two dies that use a common chip io pins.

Support for this dual chip package and chip selection scheme will need to happen upstream, this is not an smc driver issue.

closed 2019-12-15 by stale[bot]