openbmc/openbmc #3542: linux-nuvoton_git.bb fails to compile with latest master

geissonator, 2019-05-10:
Was trying to add gsj system support to CI so I put up https://gerrit.openbmc-project.xyz/c/openbmc/openbmc-build-scripts/+/21434 to add support to the CI scripts.

I hit the following error though while running this on my ppc64le system.

The key error msg seems to be "No rule to make target 'arch/arm/boot/dts/nuvoton-npcm730-gsj.dtb"

openbmc/openbmc commit is 176ed66ee047ec0a2fc6b835dc38b3815eb7d54e

```
|   AS      arch/arm/boot/compressed/piggy.o
|   LD      arch/arm/boot/compressed/vmlinux
|   OBJCOPY arch/arm/boot/zImage
|   Kernel: arch/arm/boot/zImage is ready
| NOTE: make -j128 HOSTCC=gcc  -isystem/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,-O1 HOSTCPP=gcc  -E uImage CC=arm-openbmc-linux-gnueabi-gcc   -fuse-ld=bfd -fmacro-prefix-map=/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0=/usr/src/debug/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0                      -fdebug-prefix-map=/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0=/usr/src/debug/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0                      -fdebug-prefix-map=/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot=                      -fdebug-prefix-map=/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native=  -fdebug-prefix-map=/tmp/openbmc/work-shared/gsj/kernel-source=/usr/src/kernel   LD=arm-openbmc-linux-gnueabi-ld.bfd   UIMAGE_LOADADDR=0x00008000
|   GEN     Makefile
|   Using /tmp/openbmc/work-shared/gsj/kernel-source as source for kernel
|   CALL    /tmp/openbmc/work-shared/gsj/kernel-source/scripts/checksyscalls.sh
|   CHK     include/generated/compile.h
|   Kernel: arch/arm/boot/Image is ready
|   Kernel: arch/arm/boot/zImage is ready
|   UIMAGE  arch/arm/boot/uImage
| Image Name:   Linux-5.0.13-8540490ed654f0c7992
| Created:      Fri May 10 18:33:15 2019
| Image Type:   ARM Linux Kernel Image (uncompressed)
| Data Size:    2890520 Bytes = 2822.77 KiB = 2.76 MiB
| Load Address: 00008000
| Entry Point:  00008000
|   Kernel: arch/arm/boot/uImage is ready
| NOTE: make -j128 HOSTCC=gcc  -isystem/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/recipe-sysroot-native/lib                         -Wl,-O1 HOSTCPP=gcc  -E nuvoton-npcm730-gsj.dtb
|   GEN     Makefile
| scripts/kconfig/conf  --syncconfig Kconfig
|   Using /tmp/openbmc/work-shared/gsj/kernel-source as source for kernel
| make[3]: *** No rule to make target 'arch/arm/boot/dts/nuvoton-npcm730-gsj.dtb'.  Stop.
| /tmp/openbmc/work-shared/gsj/kernel-source/Makefile:1242: recipe for target 'nuvoton-npcm730-gsj.dtb' failed
| make[2]: *** [nuvoton-npcm730-gsj.dtb] Error 2
| Makefile:173: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:15: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/linux-nuvoton/5.0.13+gitAUTOINC+8540490ed6-r0/temp/log.do_compile.118654)
ERROR: Task (/home/fspcibld/117743871/openbmc/meta-nuvoton/recipes-kernel/linux/linux-nuvoton_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2960 tasks of which 1339 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/fspcibld/117743871/openbmc/meta-nuvoton/recipes-kernel/linux/linux-nuvoton_git.bb:do_compile
Summary: There were 5 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

```

BenjaminFair, 2019-05-10:
Looks like the device tree hasn't been merged into openbmc/linux yet.

Here's the latest patch series: https://patchwork.ozlabs.org/project/openbmc/list/?series=102466

geissonator, 2019-05-13:
@shenki what's the next steps to getting the above patches into the openbmc linux kernel?

shenki, 2019-05-14:
The patches need to be reviewed. I posted some comments today. A fundamental issue is the dtb was not in the makefile.

geissonator, 2019-06-13:
Tried this again today, getting a new error in uboot recipe:
```
| ERROR: Function failed: do_compile (log file is located at /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/temp/log.do_compile.103406)
ERROR: Task (/home/fspcibld/854416885/openbmc/meta-nuvoton/recipes-bsp/u-boot/u-boot-nuvoton_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3020 tasks of which 1299 didn't need to be rerun and 1 failed.
```

```
|   gcc  -isystem/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/lib                         -L/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wp,-MD,tools/lib/.crc32.o.d -Itools -Wall -Wstrict-prototypes -O2 -fomit-frame-pointer -DCONFIG_FIT_SIGNATURE -include /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/libfdt_env.h -idirafterinclude -idirafter/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include -idirafter/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/arch/arm/include   -I/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/lib/libfdt   -I/tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/tools -DCONFIG_SYS_TEXT_BASE=0x8200 -DUSE_HOSTCC -D__KERNEL_STRICT_NAMES -D_GNU_SOURCE -pedantic -c -o tools/lib/crc32.o tools/lib/crc32.c
| In file included from /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/libfdt_env.h:12:0,
|                  from <command-line>:0:
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/include/libfdt_env.h:91:24: error: expected ')' before 'x'
|  static inline uint32_t fdt32_to_cpu(fdt32_t x)
|                         ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:9: error: expected ')' before '&' token
|   ((((x) & 0xff000000) >> 24) | \
|          ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:23: error: expected ')' before '>>' token
|   ((((x) & 0xff000000) >> 24) | \
|                        ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:30: error: expected ')' before '|' token
|   ((((x) & 0xff000000) >> 24) | \
|                               ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/include/libfdt_env.h:95:23: error: expected ')' before 'x'
|  static inline fdt32_t cpu_to_fdt32(uint32_t x)
|                        ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:9: error: expected ')' before '&' token
|   ((((x) & 0xff000000) >> 24) | \
|          ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:23: error: expected ')' before '>>' token
|   ((((x) & 0xff000000) >> 24) | \
|                        ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/git/include/compiler.h:66:30: error: expected ')' before '|' token
|   ((((x) & 0xff000000) >> 24) | \
|                               ^
| /tmp/openbmc/work/gsj-openbmc-linux-gnueabi/u-boot-nuvoton/git+npcm7xx+-r0/recipe-sysroot-native/usr/include/libfdt_env.h:100:24: error: expected ')' before 'x'
|  static inline uint64_t fdt64_to_cpu(fdt64_t x)

```

BenjaminFair, 2019-06-14:
That issue will be fixed by moving to 2019-based u-boot: https://gerrit.openbmc-project.xyz/c/openbmc/meta-nuvoton/+/22556

shenki, 2019-12-18:
GSJ was added to CI several months ago.

closed 2019-12-18 by shenki