openbmc/openbmc #3543: DHCP Hostname (transient hostname) is not getting set

aminaminf, 2019-05-15:
On our board leveraging openbmc running on an AST2520, the dhcp offered name is not  getting set due to a permission denied.  Here is the what we get in journal when /etc/hostname is empty so the prompt and transient names are set to localhost.

Anybody could shed light on how this could be fixed (polkit rule?).  

Is this working fine on  upstream openbmc based machines such as romolus, whiterspoon,etc?

Thanks,
Amin

```
root@localhost:~# systemctl --version
systemd 239
+PAM -AUDIT -SELINUX -IMA -APPARMOR -SMACK +SYSVINIT -UTMP -LIBCRYPTSETUP -GCRYPT -GNUTLS +ACL +XZ -LZ4 -SECCOMP +BLKID -ELFUTILS +KMOD -IDN2 -IDN -

PCRE2 default-hierarchy=hybridroot@localhost:~# journalctl --no-pager --no-hostname  -u systemd-networkd.service -u xyz.openbmc_project.Network.service
-- Logs begin at Wed 2019-05-15 07:43:16 UTC, end at Wed 2019-05-15 08:21:01 UTC. --
May 15 07:45:48 systemd[1]: Starting Network Service...
May 15 07:45:50 systemd-networkd[1032]: Enumeration completed
May 15 07:45:50 systemd[1]: Started Network Service.
May 15 07:45:51 systemd-networkd[1032]: request_name_destroy_callback n_ref=1
May 15 07:45:51 systemd-networkd[1032]: lo: Link is not managed by us
May 15 07:45:51 systemd-networkd[1032]: sit0: Link is not managed by us
May 15 07:45:55 systemd-networkd[1032]: eth0: Gained carrier
May 15 07:45:56 systemd-networkd[1032]: eth0: DHCPv4 address a.b.c.d/xx via a1.b1.c1.d1
May 15 07:45:57 systemd-networkd[1032]: eth0: Gained IPv6LL
May 15 07:46:00 systemd-networkd[1032]: eth0: Configured
May 15 07:46:00 systemd-networkd[1032]: Could not set hostname: Permission denied
```

aminaminf, 2019-06-04:
I appreciate it if anybody could help on how to configure openbmc to get and set hostname from DHCP server.

closed 2020-01-03 by stale[bot]

cherusk, 2020-07-20:
Persists on debian 10.

trentbuck, 2021-09-22:
Short version: you need `polkit-1`.

Hi, I am a nosy bystander.
I had the same issue (`localhost systemd-networkd[285]: Could not set hostname: Permission denied`) in a minimal Debian 11 Bullseye image (i.e. I am *not* using OpenBMC).
I fixed it by including "`polkit-1`" package in my image.
This is needed because systemd-networkd asks systemd-hostnamed to set the hostname, and hostnamed needs to ask polkitd whether it should action that request.
I did this with `APT::Install-Recommends=false`.

trentbuck, 2023-07-20:
Minor update: `polkit-1` was broken up into `polkitd` and `pkexec`.  You should only need the former.  The latter is just a convenience for attackers performing a privilege escalation.