openbmc/openbmc #3544: reported bmc state does not come back to ready after bmc reboot

kyle-ibm, 2019-05-17:
on a romulus with the latest openbmc build, the bmc state does never goes back to ready state after a bmc reboot when the system is in the power on state.

bmc version
----
    root@romulus:~# cat /etc/os-release 
    ID="openbmc-phosphor"
    NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
    VERSION="2.7.0-dev"
    VERSION_ID="2.7.0-dev-658-gc342db3"
    PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
    BUILD_ID="2.7.0-dev"
    OPENBMC_TARGET_MACHINE="romulus"
----

test procedures
----
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ./openbmctool.py -H rom1 -U root -P 0penBmc chassis power on
    Attempting login...
    Attempting to Power on...:
    {
      "data": null,
      "message": "200 OK",
      "status": "ok"
    }
    User root has been logged out
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ./openbmctool.py -H rom1 -U root -P 0penBmc chassis power status
    Attempting login...
    Chassis Power State: On
    Host Power State: Running
    BMC Power State: Ready
    User root has been logged out
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ./openbmctool.py -H rom1 -U root -P 0penBmc mc reset cold
    Attempting login...

    Attempting to reboot the BMC...:
    {
      "data": null,
      "message": "200 OK",
      "status": "ok"
    }
    User root has been logged out
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ping rom1
    PING rom1 (192.168.1.199) 56(84) bytes of data.
    From 192.168.1.33 icmp_seq=10 Destination Host Unreachable
    From 192.168.1.33 icmp_seq=11 Destination Host Unreachable
    From 192.168.1.33 icmp_seq=12 Destination Host Unreachable
    ^C
    --- rom1 ping statistics ---
    14 packets transmitted, 0 received, +3 errors, 100% packet loss, time 13287ms
    pipe 4
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ping rom1
    PING rom1 (192.168.1.199) 56(84) bytes of data.
    64 bytes from rom1 (192.168.1.199): icmp_seq=1 ttl=64 time=0.448 ms
    64 bytes from rom1 (192.168.1.199): icmp_seq=2 ttl=64 time=0.426 ms
    64 bytes from rom1 (192.168.1.199): icmp_seq=3 ttl=64 time=0.426 ms
    ^[c64 bytes from rom1 (192.168.1.199): icmp_seq=4 ttl=64 time=0.868 ms
    ^C
    --- rom1 ping statistics ---
    4 packets transmitted, 4 received, 0% packet loss, time 3075ms
    rtt min/avg/max/mdev = 0.426/0.542/0.868/0.188 ms
    kloh@openpowerlcb:~/openbmc-tools/thalerj$ ./openbmctool.py -H rom1 -U root -P 0penBmc chassis power status
    Attempting login...
    Chassis Power State: On
    Host Power State: Running
    BMC Power State: NotReady
    User root has been logged out

----

This issue causes the BMC reboot tests in the op-test-framework to fail

geissonator, 2019-05-17:
Some service must bet getting stuck. Can you attach a "journactl" output?  Looks like you're rebooting the BMC with the host up and running?

kyle-ibm, 2019-05-17:
correct. host is up already and rebooting the bmc. this doesn't seem to happen if the system is in power off state

  here is the journalctl log

----
    -- Logs begin at Fri 2019-05-17 07:58:19 UTC, end at Fri 2019-05-17 08:19:25 UTC. --
    May 17 07:58:19 romulus kernel: Booting Linux on physical CPU 0x0
    May 17 07:58:19 romulus kernel: Linux version 5.1.1-60926c8a1d00a056568cb44da9e96d3ee5f6e654 (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 Fri May 17 01:31:14 UTC 2019
    May 17 07:58:19 romulus kernel: CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
    May 17 07:58:19 romulus kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
    May 17 07:58:19 romulus kernel: OF: fdt: Machine model: Romulus BMC
    May 17 07:58:19 romulus kernel: Memory policy: Data cache writeback
    May 17 07:58:19 romulus kernel: Reserved memory: created CMA memory pool at 0x9d000000, size 16 MiB
    May 17 07:58:19 romulus kernel: OF: reserved mem: initialized node framebuffer, compatible id shared-dma-pool
    May 17 07:58:19 romulus kernel: Reserved memory: created CMA memory pool at 0x96000000, size 32 MiB
    May 17 07:58:19 romulus kernel: OF: reserved mem: initialized node jpegbuffer, compatible id shared-dma-pool
    May 17 07:58:19 romulus kernel: cma: Reserved 16 MiB at 0x9c000000
    May 17 07:58:19 romulus kernel: random: get_random_bytes called from start_kernel+0x88/0x4c4 with crng_init=0
    May 17 07:58:19 romulus kernel: Built 1 zonelists, mobility grouping on.  Total pages: 109346
    May 17 07:58:20 romulus kernel: Kernel command line: console=ttyS4,115200n8 root=/dev/ram rw
    May 17 07:58:20 romulus kernel: Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
    May 17 07:58:20 romulus kernel: Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
    May 17 07:58:20 romulus kernel: Memory: 360376K/441344K available (6144K kernel code, 390K rwdata, 1344K rodata, 1024K init, 146K bss, 15432K reserved, 65536K cma-reserved)
    May 17 07:58:20 romulus kernel: ftrace: allocating 20956 entries in 41 pages
    May 17 07:58:20 romulus kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
    May 17 07:58:20 romulus kernel: i2c controller registered, irq 16
    May 17 07:58:20 romulus kernel: clocksource: FTTMR010-TIMER2: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 77222644334 ns
    May 17 07:58:20 romulus kernel: sched_clock: 32 bits at 24MHz, resolution 40ns, wraps every 86767015915ns
    May 17 07:58:20 romulus kernel: Switching to timer-based delay loop, resolution 40ns
    May 17 07:58:20 romulus kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 49.50 BogoMIPS (lpj=247500)
    May 17 07:58:20 romulus kernel: pid_max: default: 32768 minimum: 301
    May 17 07:58:20 romulus kernel: Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
    May 17 07:58:20 romulus kernel: Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
    May 17 07:58:20 romulus kernel: *** VALIDATE proc ***
    May 17 07:58:20 romulus kernel: *** VALIDATE cgroup1 ***
    May 17 07:58:20 romulus kernel: *** VALIDATE cgroup2 ***
    May 17 07:58:20 romulus kernel: CPU: Testing write buffer coherency: ok
    May 17 07:58:20 romulus kernel: Setting up static identity map for 0x80100000 - 0x80100038
    May 17 07:58:20 romulus kernel: devtmpfs: initialized
    May 17 07:58:20 romulus kernel: clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
    May 17 07:58:20 romulus kernel: futex hash table entries: 256 (order: -1, 3072 bytes)
    May 17 07:58:20 romulus kernel: pinctrl core: initialized pinctrl subsystem
    May 17 07:58:20 romulus kernel: NET: Registered protocol family 16
    May 17 07:58:20 romulus kernel: DMA: preallocated 256 KiB pool for atomic coherent allocations
    May 17 07:58:20 romulus kernel: hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
    May 17 07:58:20 romulus kernel: hw-breakpoint: maximum watchpoint size is 4 bytes.
    May 17 07:58:20 romulus kernel: usbcore: registered new interface driver usbfs
    May 17 07:58:20 romulus kernel: usbcore: registered new interface driver hub
    May 17 07:58:20 romulus kernel: usbcore: registered new device driver usb
    May 17 07:58:20 romulus kernel: videodev: Linux video capture interface: v2.00
    May 17 07:58:20 romulus kernel: clocksource: Switched to clocksource FTTMR010-TIMER2
    May 17 07:58:20 romulus kernel: NET: Registered protocol family 2
    May 17 07:58:20 romulus kernel: tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
    May 17 07:58:20 romulus kernel: TCP established hash table entries: 4096 (order: 2, 16384 bytes)
    May 17 07:58:20 romulus kernel: TCP bind hash table entries: 4096 (order: 2, 16384 bytes)
    May 17 07:58:20 romulus kernel: TCP: Hash tables configured (established 4096 bind 4096)
    May 17 07:58:20 romulus kernel: UDP hash table entries: 256 (order: 0, 4096 bytes)
    May 17 07:58:20 romulus kernel: UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
    May 17 07:58:20 romulus kernel: NET: Registered protocol family 1
    May 17 07:58:20 romulus kernel: Unpacking initramfs...
    May 17 07:58:20 romulus kernel: Freeing initrd memory: 1492K
    May 17 07:58:20 romulus kernel: workingset: timestamp_bits=30 max_order=17 bucket_order=0
[... cut ...]
    May 17 08:11:34 romulus systemd[1]: dropbear@0-192.168.1.199:22-192.168.1.33:43354.service: Succeeded.
    May 17 08:13:13 romulus systemd[1]: Starting Cleanup of Temporary Directories...
    May 17 08:13:13 romulus systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
    May 17 08:13:13 romulus systemd[1]: Started Cleanup of Temporary Directories.
    May 17 08:19:23 romulus systemd[1]: dropbear@1-192.168.1.199:22-192.168.1.33:43356.service: Succeeded.
    May 17 08:19:25 romulus systemd[1]: Started SSH Per-Connection Server (192.168.1.33:43402).
----

mine260309, 2019-05-21:
Reproduced on Romulus, I will take a look.

mine260309, 2019-05-21:
Looks like `hostboot-settings.service` and `phosphor-cooling-type@0.service` are not started or failed.

mine260309, 2019-05-24:
* When host is not running, reboot BMC is OK, both `hostboot-settings.service` and `phosphor-cooling-type@0.service` run once and succeed.
   ```
   May 23 08:30:50 romulus systemd[1]: hostboot-settings.service: Succeeded.
   May 23 08:31:01 romulus systemd[1]: phosphor-cooling-type@0.service: Succeeded.
   ```
* When host is running and reboot BMC, it is weird that systemd is trying to start the above services for multiple times and eventually it fails due to `start-limit-hit`
   ```
   May 24 03:00:51 romulus systemd[1]: phosphor-cooling-type@0.service: Succeeded.
   May 24 03:00:55 romulus systemd[1]: phosphor-cooling-type@0.service: Succeeded.
   May 24 03:01:00 romulus systemd[1]: phosphor-cooling-type@0.service: Start request repeated too quickly.
   May 24 03:01:00 romulus systemd[1]: phosphor-cooling-type@0.service: Failed with result 'start-limit-hit'.
   May 24 03:01:00 romulus systemd[1]: Failed to start Phosphor Cooling Type.
   ```
   The same for `hostboot-settings.service`

mine260309, 2019-05-24:
Add `RemainAfterExit=true` fixes the problem, but I do not know why there is a difference between host is on and off.

mine260309, 2019-05-24:
A possible reason why the above weird behavior occurs is:
1. When host is running, there are services to check the chassis and host status, and start the related target, e.g.
   * op-reset-chassis-on@.service
   * phosphor-reset-host-running@.service
   And since the host is running, the above services will start the targets, e.g. `obmc-host-start@0.target`
2. The `obmc-host-start@0.target` has below in its unit:
   ```
   Wants=multi-user.target
   After=multi-user.target
   ```
   So systemd will try to start `multi-user.target` before `obmc-host-start@0.target`
3. Due to the `hostboot-settings.service` and `phosphor-cooling-type@0.service` do not have `RemainAfterExit=true`, systemd will treat the service not started, and start them again;
4. There are multiple services in the above case to start `multi-user.target`, and thus the `hostboot-settings.service` and `phosphor-cooling-type@0.service` will be started for multiple times, and get `start-limit-hit` failure.

So the fix to add `RemainAfterExit=true` should be correct.

mine260309, 2019-05-24:
Fixes are submitted to:
https://gerrit.openbmc-project.xyz/c/openbmc/meta-ibm/+/21933
https://gerrit.openbmc-project.xyz/c/openbmc/meta-openpower/+/21934

rfrandse, 2019-05-24:
<https://gerrit.openbmc-project.xyz/c/openbmc/meta-openpower/+/21934> Set RemainAfterExit for hostboot-settings service
Resolves  #3544 reported bmc state does not come back to ready after bmc reboot

geissonator, 2019-05-24:
I've seen this before, where oneshot services are restarted multiple times that are in the multi-user.target. I couldn't get to the bottom of what was causing it and ended up doing the same type of thing (making those services RemainAfterExit). Seems like these oneshot type services can end up getting started multiple times due to dependencies among all the multi-user target services.

closed 2019-05-28 by bradbishop