openbmc/openbmc #3545: WebUI allows same firmware to be uploaded but will not display it

JesusPhish, 2019-05-17:
The web-ui will allow a user to upload the same firmware as the one that is currently running. It will display the message that the upload was successful, the image can be found in /tmp/images, but the page does not display it. 

This could be seen as a point of confusion for end users since it appears nothing has gone wrong with the upload but the image cannot be seen on the web-ui. 

If this is the desired behaviour, maybe an enhancement with a message to inform the user that the image is already running and doesn't need to be reflashed would be useful.

mine260309, 2019-05-21:
@anoo1 I believe this is better to be fixed in bmc code manager, what do you think?

anoo1, 2019-05-23:
There's already code in the REST server(s) to return an error: https://github.com/openbmc/bmcweb/blob/4f9a213046d031b50d162f2ffb41464ab432b144/include/image_upload.hpp#L56

```            {"data",
             {{"description",
               "Version already exists or failed to be extracted"}}},
            {"message", "400 Bad Request"},
            {"status", "error"}};
```

We need to check why the GUI is not displaying the error.

gtmills, 2019-05-29:
I see this issue as well. The GUI should display an error and I think it should display the "Version already exists or failed to be extracted" error. FYI @derick-montague @jandraa

closed 2019-12-28 by stale[bot]