openbmc/openbmc #3547: Redfish AccountService Roles not matching with Redfish standard

sivassrr, 2019-05-22:
**BMC Details:**
 ```curl -k -H "X-Auth-Token: $bmc_token" -X  GET https://${bmc}/redfish/v1/Managers/bmc
{
  "@odata.context": "/redfish/v1/$metadata#Manager.Manager",
  "@odata.id": "/redfish/v1/Managers/bmc",
  "@odata.type": "#Manager.v1_3_0.Manager",
  "Actions": {
    "#Manager.Reset": {
      "ResetType@Redfish.AllowableValues": [
        "GracefulRestart"
      ],
      "target": "/redfish/v1/Managers/bmc/Actions/Manager.Reset"
    }
  },
  "DateTime": "2019-05-22T09:39:50+00:00",
  "Description": "Baseboard Management Controller",
  "EthernetInterfaces": {
    "@odata.id": "/redfish/v1/Managers/bmc/EthernetInterfaces"
  },
  "FirmwareVersion": "2.7.0-dev-695-g74c2b1336",
  "Id": "bmc",
  "Links": {
    "ManagerForServers": [
      {
        "@odata.id": "/redfish/v1/Systems/system"
      }
    ],
    "ManagerForServers@odata.count": 1
  },
  "LogServices": {
    "@odata.id": "/redfish/v1/Managers/bmc/LogServices"
  },
  "ManagerType": "BMC",
  "Model": "OpenBmc",
  "Name": "OpenBmc Manager",
  "NetworkProtocol": {
    "@odata.id": "/redfish/v1/Managers/bmc/NetworkProtocol"
  },
  "Oem": {
    "@odata.context": "/redfish/v1/$metadata#OemManager.Oem",
    "@odata.id": "/redfish/v1/Managers/bmc#/Oem",
    "@odata.type": "#OemManager.Oem",
    "OpenBmc": {
      "@odata.context": "/redfish/v1/$metadata#OemManager.OpenBmc",
      "@odata.id": "/redfish/v1/Managers/bmc#/Oem/OpenBmc",
      "@odata.type": "#OemManager.OpenBmc"
    }
  },
  "PowerState": "On",
  "SerialConsole": {
    "ConnectTypesSupported": [
      "IPMI",
      "SSH"
    ],
    "ServiceEnabled": true
  },
  "ServiceEntryPointUUID": "101e88ba-a899-4310-a03d-b5f12cdceffa",
  "Status": {
    "Health": "OK",
    "State": "Enabled"
  },
  "UUID": "49727d44-9168-4467-8256-ffb29ca2eb72"
}  ```
=========== 

**Problem Description:**

AccountService Roles are listed with "Administrator", "Operator", "User" and "Callback".
But it is not matching with Redfish standard https://www.dmtf.org/sites/default/files/standards/documents/DSP2046_2018.2_0.pdf
and also not matching with mockups: https://redfish.dmtf.org/redfish/mockups/v1/893 

openbmc related output as follows: 
```
curl -k -H "X-Auth-Token: $bmc_token" -X  GET https://${bmc}/redfish/v1/AccountService
{
  "@odata.context": "/redfish/v1/$metadata#AccountService.AccountService",
  "@odata.id": "/redfish/v1/AccountService",
  "@odata.type": "#AccountService.v1_1_0.AccountService",
  "AccountLockoutDuration": 0,
  "AccountLockoutThreshold": 0,
  "Accounts": {
    "@odata.id": "/redfish/v1/AccountService/Accounts"
  },
  "Description": "Account Service",
  "Id": "AccountService",
  "MaxPasswordLength": 20,
  "MinPasswordLength": 8,
  "Name": "Account Service",
  "Roles": {
    "@odata.id": "/redfish/v1/AccountService/Roles"
  },
  "ServiceEnabled": true
}Sivass-MacBook-Pro:~ sivas$ curl -k -H "X-Auth-Token: $bmc_token" -X  GET https://${bmc}/redfish/v1/AccountService/Roles
{
  "@odata.context": "/redfish/v1/$metadata#RoleCollection.RoleCollection",
  "@odata.id": "/redfish/v1/AccountService/Roles",
  "@odata.type": "#RoleCollection.RoleCollection",
  "Description": "BMC User Roles",
  "Members": [
    {
      "@odata.id": "/redfish/v1/AccountService/Roles/Administrator"
    },
    {
      "@odata.id": "/redfish/v1/AccountService/Roles/Operator"
    },
    {
      "@odata.id": "/redfish/v1/AccountService/Roles/User"
    },
    {
      "@odata.id": "/redfish/v1/AccountService/Roles/Callback"
    }
  ],
  "Members@odata.count": 4,
  "Name": "Roles Collection"
}Sivass-MacBook-Pro:~ sivas$ curl -k -H "X-Auth-Token: $bmc_token" -X  GET https://${bmc}/redfish/v1/Managers/bmc
{
  "@odata.context": "/redfish/v1/$metadata#Manager.Manager",
  "@odata.id": "/redfish/v1/Managers/bmc",
  "@odata.type": "#Manager.v1_3_0.Manager",
  "Actions": {
    "#Manager.Reset": {
      "ResetType@Redfish.AllowableValues": [
        "GracefulRestart"
      ],
      "target": "/redfish/v1/Managers/bmc/Actions/Manager.Reset"
    }
  },
  "DateTime": "2019-05-22T09:39:50+00:00",
  "Description": "Baseboard Management Controller",
  "EthernetInterfaces": {
    "@odata.id": "/redfish/v1/Managers/bmc/EthernetInterfaces"
  },
  "FirmwareVersion": "2.7.0-dev-695-g74c2b1336",
  "Id": "bmc",
  "Links": {
    "ManagerForServers": [
      {
        "@odata.id": "/redfish/v1/Systems/system"
      }
    ],
    "ManagerForServers@odata.count": 1
  },
  "LogServices": {
    "@odata.id": "/redfish/v1/Managers/bmc/LogServices"
  },
  "ManagerType": "BMC",
  "Model": "OpenBmc",
  "Name": "OpenBmc Manager",
  "NetworkProtocol": {
    "@odata.id": "/redfish/v1/Managers/bmc/NetworkProtocol"
  },
  "Oem": {
    "@odata.context": "/redfish/v1/$metadata#OemManager.Oem",
    "@odata.id": "/redfish/v1/Managers/bmc#/Oem",
    "@odata.type": "#OemManager.Oem",
    "OpenBmc": {
      "@odata.context": "/redfish/v1/$metadata#OemManager.OpenBmc",
      "@odata.id": "/redfish/v1/Managers/bmc#/Oem/OpenBmc",
      "@odata.type": "#OemManager.OpenBmc"
    }
  },
  "PowerState": "On",
  "SerialConsole": {
    "ConnectTypesSupported": [
      "IPMI",
      "SSH"
    ],
    "ServiceEnabled": true
  },
  "ServiceEntryPointUUID": "101e88ba-a899-4310-a03d-b5f12cdceffa",
  "Status": {
    "Health": "OK",
    "State": "Enabled"
  },
  "UUID": "49727d44-9168-4467-8256-ffb29ca2eb72"
}```

sivassrr, 2019-05-22:
Thank you @ratagupt and @Rajeswaran ThillaiGovindan for your inputs.

gtmills, 2019-06-04:
Hi @sivassrr, 

There isn't a requirement on {RoleId},  `/redfish/v1/AccountService/Roles/{RoleId}` , to only be certain IDs.
OpenBMC has defined their RoleIds as "Administrator", "Operator", "User" and "Callback". 

http://redfish.dmtf.org/schemas/v1/Role.json#/definitions/Role
http://redfish.dmtf.org/schemas/v1/RoleCollection.json#/definitions/RoleCollection

The Mockup and Redfish Resource and Schema Guide (PDF listed) are examples. 

Thanks,
Gunnar

sivassrr, 2019-06-06:
Thank you @gtmills  for your response. 

Role "User / Callback" misleading, we dont know exactly their privileges by reading the role itself.
We  have to check "/redfish/v1/AccountService/Roles/User" or "/redfish/v1/AccountService/Roles/Callback"

 Whereas if you have "ReadOnly" then it is clearly indicates that user will have only "ReadOnly". 
Same for "Operator" and "Administrator". 

Hope it is making sense.

gtmills, 2019-06-06:
@sivassrr I feel this defect has pivoted. :) 
Are you proposing we rename "CallBack" "ReadOnly" ?

@rthomaiy @ratagupt can you weight in here? 

```
bash-4.1$ curl  -k https://${bmc}/redfish/v1/AccountService/Roles/Operator
{
  "@odata.context": "/redfish/v1/$metadata#Role.Role",
  "@odata.id": "/redfish/v1/AccountService/Roles/Operator",
  "@odata.type": "#Role.v1_0_2.Role",
  "AssignedPrivileges": [
    "Login",
    "ConfigureSelf",
    "ConfigureComponents"
  ],

```
```
 curl  -k https://${bmc}/redfish/v1/AccountService/Roles/User
{
  "@odata.context": "/redfish/v1/$metadata#Role.Role",
  "@odata.id": "/redfish/v1/AccountService/Roles/User",
  "@odata.type": "#Role.v1_0_2.Role",
  "AssignedPrivileges": [
    "Login",
    "ConfigureSelf"
  ],
```
```
curl  -k https://${bmc}/redfish/v1/AccountService/Roles/Callback
{
  "@odata.context": "/redfish/v1/$metadata#Role.Role",
  "@odata.id": "/redfish/v1/AccountService/Roles/Callback",
  "@odata.type": "#Role.v1_0_2.Role",
  "AssignedPrivileges": [
    "Login"
  ],
```

ratagupt, 2019-09-30:
@sivassrr I don't think so we need an issue here.
We are aligning with redfish spec. @gtmills thanks for following it up.

sivassrr, 2019-09-30:
Ok, I agree, We will push for Callback.

closed 2019-09-30 by sivassrr

rthomaiy, 2019-10-18:
1. Just to make sure we are on the same page, Planning to get rid of callback from OpenBMC roles, will push out the updated user management document for the same.
2. There is review in flight, which changes, User to ReadOnly https://gerrit.openbmc-project.xyz/#/c/openbmc/bmcweb/+/26156/

gtmills, 2019-10-18:
I like this plan. I think this is a lot more clear.