openbmc/openbmc #3549: [Performance] Access to I2C EEPROM (at24) excruciatingly slow

madscientist159, 2019-05-22:
We've been trialling the possibility of adding a second planar VPD EEPROM accessible from the BMC, but in initial tests access via the AST2500 kernel at24 driver is excruciatingly slow.  Not only does a write cycle take nearly 10 minutes, but the host is practically hung during the process.  This results in any new SSH connections timing out, etc. during the write process.

I do note that the eeprog tool shows the same problem, pointing to a potential performance issue in the AST2500 kernel i2c drivers.

This may also help explain why the BMC seems to get slower when sensors are being polled while the host is active, making it an interesting target for optimization.

geissonator, 2019-05-23:
Any thoughts on this one @shenki or @amboar ?  Is this something going through FSI to the processor? We def know we've got some limitations with our bit banging FSI.

amboar, 2019-05-24:
> Is this something going through FSI to the processor? 

It doesn't sound like it, as @madscientist159 is fingering the ASPEED I2C driver.

> This may also help explain why the BMC seems to get slower when sensors are being polled while the host is active, making it an interesting target for optimization.

@madscientist159 Can you clarify what you're talking about here? We're going to need more data than gut feeling.

madscientist159, 2019-05-24:
> It doesn't sound like it, as @madscientist159 is fingering the ASPEED I2C driver.
>
>
> @madscientist159 Can you clarify what you're talking about here? We're going to need more data than gut feeling.

Agreed.  This was a bit of a placeholder bug so that we don't lose track of it, but here's some additional detail:

Trying to write 14k of data to a AT24C256 placed on bus 4 (kernel numbering) takes 3-4 minutes with the at24 kernel driver.  During this time the BMC is nearly hung; the device responds to ping but new connections time out and existing SSH sessions are extraordinarily sluggish (approx. 9600 baud modem levels of redraw seen when using ```top```, no way to quantify further unfortunately).  Write command is the standard ```cat file > /sys/class/i2c-dev/i2c-4/device/4-0050/eeprom``` .

closed 2019-12-23 by stale[bot]

mdmillerii, 2019-12-24:
If anyone is still seeing this, we had an issue with 1-2us delays taking >20usec and finally realized our kernel was not reaching timer hires mode, and was falling back to jiffies.

You can check this by `cat /proc/timerlist` and checking `resolution`.  
If you are in hires the resolution will be `1 nsec`.

If you don't see this check that the kernel hi res timers are enabled in your config and that your timer devices are exposed in the device tree and the kernel drivers find them.