openbmc/openbmc #3550: Cannot use a root password longer than 20 characters

madscientist159, 2019-05-23:
Most attempts to set an alphanumeric root password for the BMC longer than 20 characters will fail with:

```
chpasswd: (user root) pam_chauthtok() failed, error:
Authentication token is no longer valid; new one required
chpasswd: (line 1, user root) password not changed
```

```passwd``` fails in almost the identical manner.

Reducing password length to 20 characters allows the password set to succeed, but the entropy is reduced to ~119 bits which is insufficient for certain high security applications.

geissonator, 2019-05-23:
@joseph-reynolds @rthomaiy @ratagupt - is this a known limitation? Any plans to allow larger one's?

rthomaiy, 2019-05-23:
@geissonator yes. We limit the password length to 20 characters, if the username belong to the "ipmi" group (this is as per IPMI specification). Having said that, if the user doesn't belong to "ipmi" group, then more than 20 characters will be accepted. 
e.g. We can remove root user from "ipmi" group either during build time, or dynamically (say using busctl) and then re-trying to save the password will allow it to store more than 20 characters in length. 
This is as per the design and not an issue.

madscientist159, 2019-05-23:
@rthomaiy So there is a way to set the root password of the BMC over IPMI?

I still don't like the loss of entropy, especially with the BMC connected to a public network in certain situations.  Would it be possible to allow override?

A bit of background: In particular one of our active use cases is allowing public SSH access to the BMC.  Right now this tends to be password-based as it allows early provisioning prior to use of an authorized SSH key file.  Will IPMI ever override a "passwd -l" issued as root?

nicoleconser, 2019-05-23:
Heads up – just this morning, a retail customer of ours requested that we do not truncate the root password. All of their passwords are 32-64 characters long. I can gather more user feedback around this if needed.

rthomaiy, 2019-05-24:
@madscientist159 right now, we are treating root account as regular account, but the plan is not to treat the same as regular account.
https://github.com/openbmc/docs/blob/master/user_management.md#deployment---out-of-factory
(Already documented about how a new user account has to be deployed. Based on this, our plan is to disable the root user by default (but anybody who needs it enabled can do the same, but root user will not be part of web / ipmi managable accounts, and can be used to do ssh based on the need).
Note: even if you want it now, the same can be performed, in build time, just make sure not to list root under "ipmi" group, and the password length can be more than 20 characters (only down-side, you can't use root user to login to IPMI)

rthomaiy, 2019-06-12:
can we mark this as closed as this is expected behavior due to the IPMI Specification requirement, and provided option to set more than 20 characters by removing the "ipmi" group from root user.

closed 2020-01-11 by stale[bot]