openbmc/openbmc #3554: how to open the WEB UI

michealylj1, 2019-06-03:
Ok ,sivassrr
At first can not login webui because have no bmcweb.socket in obmc system.
The nodejs-native do_compile fail because my npm have some problem
At last to login web ui again,need to appoint port number883: https://10.0.15.177:883
Used the meta-x86/meta-intel/meta-s2600wf

_Originally posted by @FrankRenn in https://github.com/openbmc/openbmc/issues/3334#issuecomment-410411986_

michealylj1, 2019-06-03:
Hi FrankRenn,
At first can not login webui because have no bmcweb.socket in obmc system.
The nodejs-native do_compile fail because my npm have some problem
At last to login web ui again,need to appoint port number883: https://10.0.15.177:883
Used the meta-x86/meta-intel/meta-s2600wf
I just meet the same issue with you when I run openbmc in the chip "meta-evb-ast2500", Please tell me
how to add nginx service in obmc system?
how to enable bmcweb.socket in obmc system?
If I need to install nodejs in obmc system, how to do ?
Thanks,
Michael

amboar, 2019-06-04:
This seems like questions for the mailing list, this isn't really a bug report or a feature request unless it's rephrased along the lines of "Enable WebUI on ast2500-evb".

michealylj1, 2019-06-04:
[image]

[image]

michealylj1, 2019-06-04:
But I can't open web GUI by the URL  https://10.2.1.200/ or https://10.2.1.200:883/

SBoylan-Ircona, 2019-06-04:
Have you included phosphor-webui in your build? 

The basic build configuration for the EVB 2500 doesn't include the phosphor-webui package, which adds the front end web-ui interface. Have a [look here at how the meta-romulus](https://github.com/openbmc/openbmc/blob/master/meta-ibm/meta-romulus/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bbappend) build includes it.

michealylj1, 2019-06-05:
It works well after I enable "phosphor-webui". Thanks.

closed 2019-06-05 by michealylj1