openbmc/openbmc #3555: Release branching and tagging

krtaylor, 2019-06-07:
For the upcoming release (2.7) and future releases of OpenBMC, release planning needs to decide on release branching and
tagging.

In 2.6, we branched and continued developing in master, backporting any problems found to both branches. There were some
problems and concerns with how this was done, especially from within the security and testing groups.

Here is a generalized timeline drawing of the two main options (both have variations), for better understanding in decision
discussions.
[image]

joseph-reynolds, 2019-06-10:
The tagging strategy is a concern from folks who build and test from the master branch.  The build process uses tags as input to the driver ID (like `git describe`); they go into the generated "/etc/os-release" file.
At the point immediately after the 2.7 release, it is important for these tags to be accurate: the 2.7 firmware images should have 2.7 tags, and the firmware images built from master should have a tag which make sense.  (But I am not sure what is best in this situation.)

joseph-reynolds, 2019-06-10:
I **almost** had a security fix for 2.6, but it was for an older release which is no longer supported.  So yes,  I did have a question about the 2.6 branch, but no longer.  I expect the release branch to be there, or to have it created as needed.

closed 2020-01-09 by stale[bot]