openbmc/openbmc #3557: Witherspoon: Redfish: Not able to update / patch individual NTP server detail

sivassrr, 2019-06-10:
**Problem 1:  Individual NTP server not allowed to update / patch:**

We could update 2 NTP servers. (Primary / Secondary). But  if by mistake entered both entries same value, We dont have option to update the individual NTP server value. 

Problem 2:  Why both NTP servers value should be same. 

 When we enter both NTP servers as same value. It took.
What is the advantage of both NTP servers having same value. Can we not validate this and make our bmc to do intelligent check?

May be user intended to change only one digit of primary server say 9.9.9.9 to 9.9.9.8 to secondary server, but might have forgot to do that. BMC can inform about what he/she did in this case and get it corrected.
```
curl -k -H "X-Auth-Token: $bmc_token" -X PATCH -D patch.txt -d '{"NTPServers": ["2.2.3.3","2.2.3.3"]}' https://${BMC_IP}/redfish/v1/Managers/bmc/NetworkProtocol
{
 "NTP": {
    "NTPServers": [
      "2.2.3.3",
      "2.2.3.3"
    ]
 }
}Sivass-MacBook-Pro:FW691934 sivas$ curl -k -H "X-Auth-Token: $bmc_token" -X GET https://${bmc}/redfish/v1/Managers/bmc/NetworkProtocol
{
 "@odata.context": "/redfish/v1/$metadata#ManagerNetworkProtocol.ManagerNetworkProtocol",
 "@odata.id": "/redfish/v1/Managers/bmc/NetworkProtocol",
 "@odata.type": "#ManagerNetworkProtocol.v1_4_0.ManagerNetworkProtocol",
 "Description": "Manager Network Service",
 "HTTPS": {
    "Certificates": {
      "@odata.id": "/redfish/v1/Managers/bmc/NetworkProtocol/HTTPS/Certificates/"
    },
    "Port": 443,
    "ProtocolEnabled": true
 },
 "HostName": "witherspoon",
 "IPMI": {
    "ProtocolEnabled": false
 },
 "Id": "NetworkProtocol",
 "NTP": {
    "NTPServers": [
      "2.2.3.3",   <=====
      "2.2.3.3"   <======
    ],
    "ProtocolEnabled": true
 },
 "Name": "Manager Network Protocol",
 "SSH": {
    "Port": 22,
    "ProtocolEnabled": true
 },
 "Status": {
    "Health": "OK",
    "HealthRollup": "OK",
    "State": "Enabled"
 }
}

```
==========================================================

Summary: We can add enough logic / validate and make BMC intelligent which can help BMC users.

gtmills, 2019-06-18:
FYI @mine260309

mine260309, 2019-06-19:
It's related to NTP settings in phosphor-networkd, better for @ratagupt to take a look.

ratagupt, 2019-06-19:
This defect is related to bmcweb. currently for the ntp servers bmcweb doesn't allow individual ntp server address.
For the same we have the following redfish forum thread 
https://redfishforum.com/thread/186/patch-request-json-arrays

@edtanous ^^^

closed 2020-01-19 by stale[bot]