openbmc/openbmc #3558: How to Enable KCS Bridgeon AST2500-evb

michealylj1, 2019-06-10:
Hi All,

We have one switch device which is  the Intel CPU. It's already opened the KCS interface to BMC chip (AST2500).
 
root@switch1:~#  dmidecode -t 38
# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 3.0 present.

Invalid entry length (16). Fixed up to 11.
Handle 0x0058, DMI type 38, 18 bytes
IPMI Device Information
        Interface Type: KCS (Keyboard Control Style)
        Specification Version: 2.0
        I2C Slave Address: 0x10
        NV Storage Device: Not Present
        Base Address: 0x0000000000000CA2 (I/O)
        Register Spacing: Successive Byte Boundaries

Please tell me how to enable the KCS bridge daemon. It need to change the device tree to enable KCS device?

Thanks,
Michael

edtanous, 2019-07-02:
What have you tried so far?  There are a couple platforms that implement KCS.  Pushin up an example of what you've attempted along with details of where you got stuck would probably help people help to provide you input.

michealylj1, 2019-07-10:
I enable KCS channel in the device tree file as:
```
--- a/arch/arm/boot/dts/aspeed-g5.dtsi
+++ b/arch/arm/boot/dts/aspeed-g5.dtsi
@@ -387,7 +387,8 @@
 						compatible = "aspeed,ast2500-kcs-bmc";
 						interrupts = <8>;
 						kcs_chan = <3>;
-						status = "disabled";
+						kcs_addr = "0xca2";
+						status = "okay";
 					};
 				};
```

I also enable the KSC bridge in the openBMC build.

```
Index: classes/obmc-phosphor-image.bbclass
===================================================================
--- classes/obmc-phosphor-image.bbclass (revision 130424)
+++ classes/obmc-phosphor-image.bbclass (working copy)
@@ -95,7 +95,7 @@
         lrzsz \
         "
 
-OBMC_IMAGE_EXTRA_INSTALL ?= ""
+OBMC_IMAGE_EXTRA_INSTALL ?= "virtual-obmc-host-ipmi-hw"
 
 # The /etc/version file is misleading and not useful.  Remove it.
 # Users should instead rely on /etc/os-release.
Index: conf/distro/include/phosphor-defaults.inc
===================================================================
--- conf/distro/include/phosphor-defaults.inc   (revision 130173)
+++ conf/distro/include/phosphor-defaults.inc   (working copy)
@@ -11,7 +11,8 @@
 PREFERRED_PROVIDER_virtual/obmc-chassis-mgmt ?= "obmc-phosphor-chassisd"
 PREFERRED_PROVIDER_virtual/obmc-fan-mgmt ?= "obmc-phosphor-fand"
 PREFERRED_PROVIDER_virtual/obmc-flash-mgmt ?= "obmc-phosphor-flashd"
-PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw ?= "phosphor-ipmi-hw-example"
+#PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw ?= "phosphor-ipmi-hw-example"
+PREFERRED_PROVIDER_virtual/obmc-host-ipmi-hw ?= "phosphor-ipmi-kcs"
 
```

michealylj1, 2019-07-10:
We also use the oscilloscope to measure the KSC signal in the LPC bus between X86 and AST2500, When I run command as "modprobe ipmi_si" in the SONIC system, It can be saw the KCS request from X86 to AST2500, but there is no any response from BCM side. So I just suspect the AST2500 don't run in KCS mode.

michealylj1, 2019-07-10:
kcsbridged is running as:

[image]

ChaoyangZhang, 2019-11-01:
How about this issue? I also have this problem.

closed 2020-06-01 by stale[bot]

OguzhanCaglar, 2022-09-08:
Hello, How can I access the SMBIOS information with the dmidecode? Dmidecode returns "No SMBIOS nor DMI entry point found, sorry.". How can I get this data?