openbmc/openbmc #3560: what is the default of  PowerState HostState and ChassisState

FrankRenn, 2019-06-13:
This three State make me confuse, what is the default , and The connection between them is what.

Thanks alot

geissonator, 2019-06-13:
OpenBMC has 3 independent objects that represent the state of, BMC, Chassis, and Host.

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/BMC.interface.yaml

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/Chassis.interface.yaml

https://github.com/openbmc/phosphor-dbus-interfaces/blob/master/xyz/openbmc_project/State/Host.interface.yaml

Each has a state it can be, and a state you can request of it.

https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/22461/1/designs/extended-host-software-state.md and https://gerrit.openbmc-project.xyz/c/openbmc/docs/+/22358/1/designs/ipmi_and_redfish_chassis_control.md provide some background on them and some coming enhancements.

systemd is used for the relationships among them, you can get more info in https://github.com/openbmc/docs/blob/master/openbmc-systemd.md

In general this question is best for the mailing list or IRC.

closed 2019-06-13 by geissonator

FrankRenn, 2019-06-14:
OK ,thanks