openbmc/openbmc #3561: x86-power-control and obmc-target conflict

SBoylan-Ircona, 2019-06-14:
I have a current problem when trying to include x86-power control in my build. I've included a snippet of the error log below. 

On my build I've copied a local version of x86-power-control into recipes-phosphor/chassis/x86-power-control

```
 volatile-binds             noarch         1.0-r0                 oe-repo 9.6 k
 x86-power-control          armv6          1.0+git0+676d2c3997-r1 oe-repo  96 k
 xz                         armv6          5.2.4-r0               oe-repo  44 k
Installing weak dependencies:
 avahi-daemon               armv6          0.7-r0                 oe-repo  53 k
 busybox-syslog             armv6          1.30.1-r0              oe-repo 9.5 k
 busybox-udhcpc             armv6          1.30.1-r0              oe-repo 8.1 k
 ca-certificates            noarch         20190110-r0            oe-repo 140 k
 e2fsprogs-e2fsck           armv6          1.44.5-r0              oe-repo 110 k
 iptables-modules           armv6          1.6.2-r0               oe-repo 8.2 k
 libgpiod-tools             armv6          1.3-r0                 oe-repo  18 k
 libnss-mdns                armv6          0.10-r7                oe-repo  16 k
 ncurses-terminfo-base      armv6          6.1+20181013-r0        oe-repo  14 k
 obmc-targets               noarch         1.0-r1                 oe-repo  20 k
 os-release                 noarch         1.0-r0                 oe-repo 6.6 k
 packagegroup-obmc-ipmid-providers-libs
                            noarch         1.0-r1                 oe-repo 5.6 k
 phosphor-state-manager-host-check
                            armv6          1.0+git0+624161cd2e-r1 oe-repo  20 k
 phosphor-state-manager-reset-sensor-states
                            armv6          1.0+git0+624161cd2e-r1 oe-repo 8.1 k
 slpd-lite                  armv6          1.0+git0+a592888328-r1 oe-repo  22 k
 systemd-compat-units       armv6          1.0-r29                oe-repo 6.2 k
 systemd-extra-utils        armv6          1:241+0+c1f8ff8d0d-r0  oe-repo  90 k

Transaction Summary
================================================================================
Install  344 Packages

Total size: 21 M
Installed size: 67 M
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Error: Transaction check error:
  file /lib/systemd/system/obmc-chassis-hard-poweroff@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-chassis-poweroff@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-chassis-poweron@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-chassis-powerreset@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-host-reboot@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-host-start@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-host-startmin@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
  file /lib/systemd/system/obmc-host-stop@.target conflicts between attempted installs of obmc-targets-1.0-r1.noarch and x86-power-control-1.0+git0+676d2c3997-r1.armv6
```

From what I can gather the issue is that x86-power-control is being built for armv6 while obmc-targets is being built for allarch (from looking at it's .bb file). The two are both trying to include several .target files of the same name into /lib/systemd/system/ and this is causing a problem.

I've tried to get x86-power-control to build for allarch and that causes other issues resulting in it being unable to build. I'm unsure how to resolve this short of modifying obmc-target to not include those files. Or should these two packages just not be included in the same project?

Does anyone have any other suggestions?

SBoylan-Ircona, 2019-06-14:
Having just found this [bbappend file ](https://github.com/openbmc/openbmc/blob/master/meta-intel/meta-common/recipes-core/systemd/obmc-targets.bbappend) that removes the files, I assume that is the only solution.

geissonator, 2019-06-14:
Didn't know about that bbappend. In general the design is to have common targets installed by one recipe and then have the required services be installed by the meta layer that provides them. Not sure why Intel needed to install the targets as well @feistjj ?

bradbishop, 2019-06-14:
> In general the design is to have common targets installed by one recipe

fwiw I’d like to see those common targets moved over to phosphor-state-manager someday.

closed 2020-01-13 by stale[bot]