openbmc/openbmc #3562: Unable to configure IP address on latest build

prkatti1, 2019-06-17:
Unable configure IP address on latest build.

cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-dev"
VERSION_ID="2.7.0-dev-880-gc9ed720bc"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
BUILD_ID="2.7.0-dev"
OPENBMC_TARGET_MACHINE="witherspoon"

```curl -k -H "X-Auth-Token: $bmc_token" -X PATCH -D patch.txt -d '{"IPv4Addresses": [{},{},{"Address": "10.7.7.7","AddressOrigin": "Static","SubnetMask": "255.255.0.0","Gateway":"10.7.7.1"}]}' https://${bmc}/redfish/v1/Managers/bmc/EthernetInterfaces/eth0
{
  "AddressOrigin@Message.ExtendedInfo": [
    {
      "@odata.type": "/redfish/v1/$metadata#Message.v1_0_0.Message",
      "Message": "The property AddressOrigin is not in the list of valid properties for the resource.",
      "MessageArgs": [
        "AddressOrigin"
      ],
      "MessageId": "Base.1.4.0.PropertyUnknown",
      "Resolution": "Remove the unknown property from the request body and resubmit the request if the operation failed.",
      "Severity": "Warning"
    }
  ]
}-```

```HTTP/1.1 400 Bad Request
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: DENY
Pragma: no-cache
Cache-Control: no-Store,no-Cache
Content-Security-Policy: default-src 'self'; img-src 'self' data:
X-XSS-Protection: 1; mode=block
X-Content-Type-Options: nosniff
X-UA-Compatible: IE=11
Content-Type: application/json
Server: iBMC
Date: Mon, 17 Jun 2019 11:23:51 GMT
Content-Length: 490
```

geissonator, 2019-06-17:
Looks like AddressOrigin is a read-only parameter and was removed from the patch command in gerrit commit https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/22501  If you disagree with that, probably best to move this bug over to bmcweb and talk with @edtanous and team.

ratagupt, 2019-06-18:
Seems we need to revert this change, as the PATCH on ipv4address is used for adding the IPaddress as well as to patch the existing IP address.

We follow the following semantics for the PATCH.

```7.6.1. PATCH on array properties
There are three possible styles of array properties in a Resource, which are detailed in the Array
properties clause.
Within a PATCH request, the service shall accept null to remove an element, and accept an empty
object {} to leave an element unchanged. Array properties using the fixed or variable length style will
remove those elements, while array properties using the rigid style will replace removed elements with
Redfish Specification DSP0266
44 Published Version 1.7.0
null elements. A service may indicate the maximum size of an array by padding null elements at the
end of the array sequence.
When processing a PATCH request, the order of operations shall be:
• Modifications
• Deletions
• Additions
A PATCH request with fewer elements than currently exist in the array shall remove the remaining
elements of the array.
For example, a fixed length style array of 'Flavors' indicates the service supports a maximum of six
elements (by padding the array with null elements), with four populated.
"Flavors": [ "Chocolate", "Vanilla", "Mango", "Strawberry", null, null ]
A client could issue the following PATCH request to remove Vanilla, replace Strawberry with
Cherry, and add Coffee and Banana to the array, while leaving the other elements unchanged.
"Flavors": [ {}, null, {}, "Cherry", "Coffee", "Banana" ]
The resulting array after the PATCH is:
"Flavors": [ "Chocolate", "Mango", "Cherry", "Coffee", "Banana", null ]```

However I feel that Ipv4address property is readonly and we should use the "IPv4StaticAddresses" to configure the Ipaddress.

NOTE:

IPV4address will used to show all the IPaddresses(Static/DHCP/ZeroConf) on the interface. PATCH operation should not be used.

IPv4StaticAddresses  will used to configure/show the static IPaddress on the interface and PATCH operation should be allowed.

.

raviteja-b, 2019-07-05:
Resolved by: https://gerrit.openbmc-project.xyz/c/openbmc/bmcweb/+/22940/

prkatti1, 2019-07-09:
Its working on latest build, so closing this issue.

closed 2019-07-09 by prkatti1