openbmc/openbmc #3563: Corrupted journal causing systemd to exceed SystemMaxUse

geissonator, 2019-06-18:
I recently got on a system which had a full persistent filesystem. After some looking around, I found that the systemd journal was using 5.24MB in /var/log/journal/. Per the SystemMax, this should have been capped at 2.5MB.

I found the following in the journal directory:
```
drwxr-sr-x    2 root     systemd-      3640 Mar 14 18:24 .
drwxr-sr-x    3 root     systemd-       256 May  2  2018 ..
-rw-r-----    1 root     systemd-         0 Mar 14 18:24 system.journal
-rw-r-----    1 root     systemd-    139264 Mar 14 21:01 system@0005841207a76cd5-c2fc28d236ce2760.journal~
-rw-r-----    1 root     systemd-     12288 Mar 14 18:30 system@0005841207b6a65f-6afd928bbd16adf7.journal~
-rw-r-----    1 root     systemd-    524288 Mar 14 18:24 system@0005841207bb809c-929491b156b363a1.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:24 system@00058412080f7266-1c090b19a36c585f.journal~
-rw-r-----    1 root     systemd-    499712 Mar 14 18:24 system@000584120819a2bc-4b3b65058637ac55.journal~
-rw-r-----    1 root     systemd-     16384 Mar 14 18:24 system@00058412089edc13-f332755b8f766bbc.journal~
-rw-r-----    1 root     systemd-     36864 Mar 14 18:24 system@00058412099a6710-f8a4930ac08ea9f3.journal~
-rw-r-----    1 root     systemd-    172032 Mar 14 18:25 system@000584120e37e9ca-3b739541c0784394.journal~
-rw-r-----    1 root     systemd-      4096 Mar 14 18:26 system@000584120e4f8708-414d79ebad14973b.journal~
-rw-r-----    1 root     systemd-    524288 Mar 14 18:27 system@0005841211f1c96f-c462e76e08c950a7.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:26 system@00058412122217dc-4e223760f7e4d9d3.journal~
-rw-r-----    1 root     systemd-     90112 Mar 14 18:27 system@0005841213916d72-3101bac0bd011f44.journal~
-rw-r-----    1 root     systemd-    278528 Mar 14 18:27 system@0005841213d86fd6-c7a5ce88dac2c6ca.journal~
-rw-r-----    1 root     systemd-     28672 Mar 14 18:28 system@0005841215c21f33-5308b94c9369d482.journal~
-rw-r-----    1 root     systemd-      8192 Mar 14 18:31 system@000584121fe35348-af5e1624b9ea7f17.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:33 system@00058412285de97c-4cba0ba59f472b5c.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:33 system@0005841228b01cb7-fbe903b9cad40ae3.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:34 system@000584122bb57055-4f0d55378e4e0b52.journal~
-rw-r-----    1 root     systemd-      8192 Mar 14 18:34 system@0005841230223c5b-e3459d02d79186c9.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:35 system@0005841235a2705d-6bd3d9456a8c42bd.journal~
-rw-r-----    1 root     systemd-       240 Mar 14 18:41 system@000584124af5867c-65356b50b33fe2ed.journal~
-rw-r-----    1 root     systemd-      4096 Mar 14 21:00 system@00058414382c6f48-7303d2f824805754.journal~
-rw-r-----    1 root     systemd-     12288 Mar 15 00:28 system@0005841721df1a17-91c7577e3b863f02.journal~
-rw-r-----    1 root     systemd-    262144 Mar 16 11:48 system@00058434bd9785cf-67230a5d04622e13.journal~
-rw-r-----    1 root     systemd-    524288 Mar 28  2019 system@8919839c885d4747aabc06d24b4b97fe-000000000002fcf0-00058527690cff46.journal
-rw-r-----    1 root     systemd-    524288 Mar 28  2019 system@8919839c885d4747aabc06d24b4b97fe-000000000002fecb-00058527b4ee6a67.journal
-rw-r-----    1 root     systemd-    524288 Mar 31  2019 system@8919839c885d4747aabc06d24b4b97fe-0000000000030133-000585288f97d115.journal
-rw-r-----    1 root     systemd-    524288 Mar 31  2019 system@8919839c885d4747aabc06d24b4b97fe-00000000000303f2-00058563be07ef21.journal
-rw-r-----    1 root     systemd-    524288 Mar 16 11:48 system@a5384ba9812e4b7ea78540be292e58b2-00000000000008f7-0005843451d8fdde.journal
```

Two issues here, something is causing corrupted journal files (the ~ on the end of the filename) and systemd does not count those files towards the SystemMaxUse setting.

We are running systemd 237 on this system so we can not open a bug on systemd (they only allow bugs on their last 2 releases). I'm opening this as an issue we can track any further recreates of this with. If we see with more recent systemd then we can open a bug.

This is a downstream build, but it's based off of v2.3-476-g2d622cb

```
journalctl | more
Journal file /var/log/journal/dbf6ec8de11447969faff1bece94f528/system.journal is truncated, ignoring file.
-- Logs begin at Thu 2019-03-14 18:24:06 UTC, end at Sun 2019-03-31 13:21:59 UTC. --
Mar 14 18:24:06 witherspoon kernel: Booting Linux on physical CPU 0x0
Mar 14 18:24:06 witherspoon kernel: Linux version 4.18.12-b953727d56c4de63eb0fad1f079f41004e61d713 (oe-user@oe-host) (gcc version 7.3.0 (GCC)) #1 Thu Mar 14 18:51:36 UTC 2019
Mar 14 18:24:06 witherspoon kernel: CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
Mar 14 18:24:06 witherspoon kernel: CPU: PIPT / VIPT nonaliasing data cache, VI
...
```
```
cat /lib/systemd/journald.conf.d/journald-size-policy.conf 
# This file overrides a default for journald
#
# - Set a maximum size for journald in filesystem
#
# See journald.conf(5) for details

[Journal]
SystemMaxUse=2.5M
```

geissonator, 2019-06-18:
One thing we noticed was that the system time for the BMC was actually earlier then the most recent journal entry. i.e. to journald it looked like the BMC had gone back in time. That may be causing some of the corruption issues.

```
date
Thu Mar 14 18:26:50 UTC 2019

journalctl -f

-- Logs begin at Thu 2019-03-14 18:24:07 UTC. --
Mar 14 18:29:42 witherspoon systemd[1]: Reached target Final Step.
Mar 14 18:29:42 witherspoon systemd[1]: Starting Reboot...
Mar 14 18:29:42 witherspoon systemd[1]: Shutting down.
Mar 14 18:29:42 witherspoon systemd[1]: Hardware watchdog 'aspeed_wdt', version 0
Mar 14 18:29:42 witherspoon kernel: watchdog: watchdog0: watchdog did not stop!
Mar 14 18:29:42 witherspoon kernel: systemd-shutdow: 63 output lines suppressed due to ratelimiting
Mar 14 18:29:42 witherspoon systemd[1]: Set hardware watchdog to 10min.
Mar 14 18:29:42 witherspoon systemd-shutdown[1]: Syncing filesystems and block devices.
Mar 14 18:29:44 witherspoon systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Mar 14 18:29:44 witherspoon systemd-journald[997]: Journal stopped
```

geissonator, 2019-06-19:
@gkeishin ran some tests against master. That has systemd 242. 

It seems to handle a time change going backwards much better:

```
Jun 19 12:59:21 wsbmc002 systemd[1]: Stopping Network Service...
Jun 19 12:59:21 wsbmc002 systemd[1]: systemd-networkd.service: Succeeded.
Jun 19 12:59:21 wsbmc002 systemd[1]: Stopped Network Service.
Jun 19 12:59:21 wsbmc002 systemd[1]: Starting Network Service...
Jun 19 12:59:21 wsbmc002 systemd-networkd[1575]: eth0: Gained IPv6LL
Jun 19 12:59:21 wsbmc002 systemd-networkd[1575]: Enumeration completed
Jun 19 12:59:21 wsbmc002 systemd-networkd[1575]: sit0: Could not acquire IPv4 link-local address: Invalid argument
Jun 19 12:59:22 wsbmc002 systemd[1]: Started Network Service.
Jun 01 12:58:43 wsbmc002 systemd-timedated[1550]: Changed local time to Sat Jun  1 12:58:43 2019
Jun 01 12:58:43 wsbmc002 phosphor-timemanager[1464]: BMC system time is changed
Jun 01 12:58:48 wsbmc002 systemd-networkd[1575]: eth0: Configured
Jun 01 12:58:52 wsbmc002 phosphor-network-manager[1338]: Refreshing the objects.
Jun 01 12:58:52 wsbmc002 phosphor-network-manager[1338]: Refreshing complete.
```

mdmillerii, 2019-06-19:
The [journal server code](https://github.com/systemd/systemd/blob/b19eab1f74f5105fc5737b3790071175ea1a9292/src/journal/journald-server.c#L100) does look at journal and journal~ files.   Can you include `du -a` or `ls -s` output to show the blocks allocated (which is what is added?   I also see references to mount space and minimums (`df`).  Or just [find the usage message](https://github.com/systemd/systemd/blob/b19eab1f74f5105fc5737b3790071175ea1a9292/src/journal/journald-server.c#L205).

closed 2020-01-19 by stale[bot]