openbmc/openbmc #3565: Not able to mount Filesystems

Pranav1711, 2019-06-25:


Pranav1711, 2019-06-25:
tunneling driver
[    3.385153] NET: Registered protocol family 17
[    3.389773] 8021q: 802.1Q VLAN Support v1.8
[    3.403277] printk: console [netcon0] enabled
[    3.407761] netconsole: network logging started
[    3.412315] hctosys: unable to open rtc device (rtc0)
[    3.424002] Freeing unused kernel memory: 1024K
[    3.431226] Checked W+X mappings: passed, no W+X pages found
[    3.437019] Run /init as init process
cat: read error: Is a directory
cat: read error: Is a directory
rofs = squashfs rwfs = jffs2
mount: mounting /dev/mtdblock on run/initramfs/ro failed: No such device
mount: mounting /dev/mtdblock on run/initramfs/rw failed: No such device

Mounting read-write /dev/mtdblock filesystem failed.  Please fix and run
       mount /dev/mtdblock run/initramfs/rw -t jffs2 -o rw
to to continue, or do change nothing to run from RAM for this boot.
Enter password to try to manually fix.
After fixing run exit to continue this script, or reboot -f to retry, or
touch /takeover and exit to become PID 1 allowing editing of this script.
Give root password for system maintenance
(or type Control-D for normal startup):

mdmillerii, 2019-06-25:
> Please let us know how we can disable Overlay use a single RW Filesystem.

You did not provide any details of how much space you have or where you would store said filesystem (or which filesystem type you would like to use).

If you have space you can change the IMAGE_TYPE to any that yocto supports and it will create a filesystem image and place it in the deploy directory.

# Background on the file system. 

The much belated file system documentation is still in progress so I will give you a short summary here.

The base solution is to use an build with just `busybox`, three shell scripts along with a few config files to find and mount the squashfs (which is highly compressed in a blockwize manor, and hence read-only) and then allow modifications to be stored persistently via a separate initially empty read-write file system mounted with overlayfs.

The supplied `init` script, `obmc-init.sh`, looks in the `/sys` filesystem to find a MTD partition labeled `rofs` that it expects to find the root `squashfs`, and similarly looks for a MTD partition labeled `rwfs` that it uses for the writable overlay.  The partition name is usually supplied by the device tree (and are supplied via the shared file `openbmc-flash-layout.dtsi`) although you could use `Das U-boot` support for passing its partitions to the Linux kernel via the kernel command line.

As the prompt indicates, you can enter it at that prompt and get a shell into the initramfs.  If you configured the default root password in your conf file it is `0penBMC`.

pranav-shivalingaiah, 2019-06-30:
Hi Thank you,
Due to few changes not implemented, we were facing the FS issue, now its resolved.
As of now we shall go ahead with Overlay.
Thank you.

anoo1, 2019-07-19:
> Due to few changes not implemented, we were facing the FS issue, now its resolved.

Good to hear. Closing.

closed 2019-07-19 by anoo1

xiongyw, 2022-02-08:
> although you could use Das U-boot support for passing its partitions to the Linux kernel via the kernel command line.

@mdmillerii [mdmillerii](https://github.com/mdmillerii) Can you please give instructions on how to do so (in u-boot shell environment, without modifying the built fit image)? 

My scenario is as below: I just got an ast2600-evb board. As tested, the default image (from ASpeed) in the 1st spi flash contains 3 partitions (`u-boot`, `u-boot-env`, and `fit`).  Before reprogramming that part of the flash memory using a newly built openbmc image, I want to first try the new image via tftp (since I want to minimize the risk of bricking the board).  After `tftp` and `boom`, I encounter the same error as shown in this post. After reading your replies, I now understand the reason (i.e., the actual content on the flash is not a squashfs yet, or, the fdt in the newly built image does not match the actual flash content).

To continue the testing (without erasing the original image), I plan to write the new image into another flash memory (e.g., the 2nd flash memory), and *somehow* provide the corresponding partition info of the 2nd flash memory to kernel (also overriding the partition info in the fdt inside the fit image, *somehow*). If this is possible and done correctly, then the kernel should be able to find the correct mtd device containing the squashfs. 

But I don't know how to provide partition info in kernel cmdline.

Thanks.