openbmc/openbmc #3566: Mount failed in dual flash chips design

MichaelWang2017, 2019-06-27:
Hi all,
We have a board that uses two BMC flash chips (flash0 uses CS0, flash1 uses CS1) and stores the same image. 
The dts partition configuration is as follows:
[image]
openbmc-flash-layout-inspur-64MB.dtsi:
[image]
alt-flash-layout-inspur-64MB.dtsi:
[image]
In addition, I modified _obmc-init.sh_ as follows:
[image]
[image]

The problem encountered is that if only one flash chip is placed, either flash0 or flash1, the following problems will be encountered:
[image]

mine260309, 2019-06-27:
You should be able to enter the password (default `0penBmc`) and get a root shell, then you should be able to see what kind of mtd devices there are.

MichaelWang2017, 2019-06-27:
Okay, Thank you @mine260309, I will check it later.
In addition, I tried to comment out the flash0 in the dts file as follows:
[image]

My understanding is that at this time, regardless of whether mount is successful or not, aspeed-smc.c should at least correctly initialize flash1, but I found that flash1 was not properly initialized:
[image]

mdmillerii, 2019-06-27:
The SMC has a configurable window for each chip select.  I'm sure the code has not been tested when skipping chip selects, and not sure what the hardware would do if chip select 0 and 1 had the same window.  I'd start by putting the first chip back in the device tree.

For some reason your JEDEC id read back as `00 00 00` (unless there are some off the right side of the picture) so the chip is not being found.

Note: be aware that the watchdog toggle will swap chipselect 0 and 1 each time it triggers, so be aware the label are dependent on the state of the watchdog.

closed 2020-01-26 by stale[bot]

mdmillerii, 2020-02-08:
Upon review after others commented, the scenerio is not supported by the driver initialization but perhaps should be.

The current linux driver for the spi interface (`drivers/mtd/spi-nor/aspeed-smc.c`) requires each described flash chip to be present and respond with jedec id when probed or will return -ENODEV for the whole controller.  (The 2500 has 3 controllers - 1 for the SOC and two for the host; the 2400 has 1 host and 1 SOC).