openbmc/openbmc #3570: Cannot devtool modify phosphor-ipmi-host with the latest version

yongli3, 2019-07-01:
Devtool modify phosphor-ipmi-host failed with the below error message:
cp: cannot stat 'entity.yaml': No such file or directory

After revert the below change, issue fixed:
https://github.com/openbmc/meta-phosphor/commit/fe88ee4211c93f4ca4acba5ecec502aef17944ce

@bradbishop  Any comments?

bradbishop, 2019-07-08:
thanks for the report @yongli3, I will figure it out.

bradbishop, 2019-08-22:
at 5:09 AM, Yong Li <notifications@github.com> wrote:

> Devtool modify phosphor-ipmi-host failed with the below error message:
> cp: cannot stat 'entity.yaml': No such file or directory
>
> After revert the below change, issue fixed:
> openbmc/meta-phosphor@fe88ee4
>
> @bradbishop Any comments?

I suspect this is going to get fixed by this patch from Patrick:

https://gerrit.openbmc-project.xyz/c/openbmc/meta-phosphor/+/24552

Apologies for never getting back to this.

yongli3, 2019-08-30:
Thanks @bradbishop for the information. I tested the change #24552, it can fix the devtool modify error. Will add comments on the code review, and wait for the merge.

closed 2020-03-30 by stale[bot]