openbmc/openbmc #3572: Unable to delete staticnameserver from BMC

prkatti1, 2019-07-02:
Unable to delete staticnameserver entry from BMC.
```curl -k -H "X-Auth-Token: $bmc_token" -X PATCH -D patch.txt -d '{"StaticNameServers": ["null","3.3.3.7"]}' https://${bmc}/redfish/v1/Managers/bmc/EthernetInterfaces/eth0
 
curl -k -H "X-Auth-Token: $bmc_token" -X GET https://$bmc/redfish/v1/Managers/bmc/EthernetInterfaces/eth0
```
```  "NameServers": [
    "null",
    "3.3.3.7"
  ],
  "SpeedMbps": 0,
  "StaticNameServers": [
    "null",
    "3.3.3.7"
  ],
```

ratagupt, 2019-07-02:
@prkatti1 Please open the issue under bmcweb repo.

prkatti1, 2019-07-02:
@ratagupt closing this, will be tracked via https://github.com/openbmc/bmcweb/issues/93

closed 2020-01-31 by stale[bot]