openbmc/openbmc #3574: rngd consume too much CPU

mine260309, 2019-07-03:
The `rngd.service` consumes almost all CPU and the BMC becomes slow.

```
# top
CPU:   6% usr  69% sys  23% nic   0% idle   0% io   0% irq   0% sirq
Load average: 1.12 1.15 1.10 2/122 5168
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  713     1 root     RN   12716   3%  87% /usr/sbin/rngd -f -r /dev/hwrng
```

BMC build:
```
# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.7.0-dev"
VERSION_ID="2.7.0-dev-1022-g934d64a"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.7.0-dev"
BUILD_ID="2.7.0-dev"
OPENBMC_TARGET_MACHINE="romulus"
```

mine260309, 2019-07-03:
A notable change is rng-tools is updated to `rng-tools_6.7.bb` from `rng-tools_6.6.bb` in subtree.

mine260309, 2019-07-04:
BMC version `2.7.0-dev-947-g9024596` does not have this issue, and it is before f3fd288 (which pulls in `rng-tools_6.7.bb`), so it is very likely that f3fd288 introduces the issue.

mine260309, 2019-07-04:
The root cause is due to the patch introduced in f3fd288.

Related original code
```
// In thread_entropy_task():
          while(written != me->buf_sz) {
              message(LOG_DAEMON|LOG_DEBUG, "Writing to pipe\n");
              ret = write(me->pipe_fd, &tmpbuf[written], me->buf_sz - written);
              message(LOG_DAEMON|LOG_DEBUG, "DONE Writing to pipe with return %ld\n", ret);
              if (first)
                  me->active = 1;
              ...
          }

// In init_jitter_entropy_source():
     /* Make sure all our threads are doing their jobs */
     for (i=0; i < num_threads; i++) {
         while (tdata[i].active == 0)
             sched_yield();
         message(LOG_DAEMON|LOG_DEBUG, "CPU Thread %d is ready\n", i);
     }
```

It works fine, that when `thread_entropy_task()` runs, it sets the `active` to 1, and `init_jitter_entropy_source()` waits for it to become 1.

However, the patch  `fix-rngd-fail-to-stop.patch` changes the behavior:
```
diff --git a/rngd_jitter.c b/rngd_jitter.c
index 54070ae..7a69bf9 100644
--- a/rngd_jitter.c
+++ b/rngd_jitter.c
@@ -280,7 +280,7 @@ static void *thread_entropy_task(void *data)

                /* Write to pipe */
                written = 0;
-               while(written != me->buf_sz) {
+               while(me->active && written != me->buf_sz) {
                        message(LOG_DAEMON|LOG_DEBUG, "Writing to pipe\n");
                        ret = write(me->pipe_fd, &tmpbuf[written], me->buf_sz - written);
                        message(LOG_DAEMON|LOG_DEBUG, "DONE Writing to pipe with return %ld\n", ret);
```

The `while` loop will never get to run due to the `active` is 0, and it never gets a chance to set it to 1, and `init_jitter_entropy_source()` is always spinning to wait for it to become 1.

We should remove the patch, and it works fine.

mdmillerii, 2019-07-08:
The patch apparently was developed response to a RH bug but upstream actually committed a different fix.

The 2.8 yocto recipe for rng-tools is the upstream HEAD, how about cherry picking?

mine260309, 2019-07-08:
Or let's disable jitter entropy for OpenBMC project, which eliminates any high CPU usage case either in the startup or idle state.

See https://lists.ozlabs.org/pipermail/openbmc/2019-July/016964.html

rfrandse, 2019-07-08:
<https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/23345> rng-tools: Disable jitter entropy
Resolves  #3574 rngd consume too much CPU

mdmillerii, 2019-07-09:
@bradbishop please pull upstream 2.8 that removed the bogus "backport" patch.

geissonator, 2019-07-31:
We're currently blocked moving to the upstream fix due to some other unrelated issues in the upstream package. Brad communicated those problems to @amboar and @wak-google in IRC.

```
[19:31:21]  <@radsquirrel>	amboar when you have a few can you look at autoconf-archive commit b5a2d69f7a ?
[19:31:53]  <@radsquirrel>	all the repos using CODE_COVERAGE_RULES break when we pick that up.
[19:37:14]  <@radsquirrel>	wak-work there are a number of packages that RDEPEND on gtest, which breaks when we pick up your upstream changes to the googletest recipe
```

Once we get these addressed in our code base we can move to the upstream fix for this issues.

bradbishop, 2019-08-16:
> @bradbishop please pull upstream 2.8 that removed the bogus "backport" patch.

done!  finally...

mine260309, 2019-08-19:
hmm, it's in https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/24527 but not merged yet...

bradbishop, 2019-08-19:
at 10:57 PM, Lei YU <notifications@github.com> wrote:

> hmm, it's in https://gerrit.openbmc-project.xyz/c/openbmc/openbmc/+/24527  
> but not merged yet…

Oops…I thought rngd was in meta-openembedded and not oe-core.  Sorry ABT.   
The oe-core update should be coming pretty soon…just a couple issues to  
work through there.

mine260309, 2019-09-18:
Upstream fixed the issue and subtree update at 96ff198 fixed this issue.

closed 2019-09-18 by mine260309