openbmc/openbmc #3575: Where can find the service or daemon call shutdown shell?

coffee2235, 2019-07-04:
Where can find the service or daemon call shutdown shell in' /run/initramfs'  when flashing image file to flash?
(I put my image-bmc file to /run/initramfs and reboot my device to flash)

shutdown: reboot --timeout 90000000us --log-level 6 --log-target kmsg --log-color 
=>from systemd

sort -r
=>from /run/initramfs/shutdown

I trace code to find out where is the log belong.
But I couldn't find where will call shutdown shell

This is my log during flash process
```
[  OK  ] Reached target Final Step.
         Starting Reboot...
[  254.344140] watchdog: watchdog0: watchdog did not stop!
[  254.379879] systemd-shutdow: 25 output lines suppressed due to ratelimiting
[  254.431239] systemd-shutdown[1]: Syncing filesystems and block devices.
[  254.446459] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[  254.477433] systemd-journald[636]: Received SIGTERM from PID 1 (systemd-shutdow).
[  254.498318] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[  254.521938] systemd-shutdown[1]: Hardware watchdog 'aspeed_wdt', version 0
[  254.535898] systemd-shutdown[1]: Unmounting file systems.
[  254.553958] [1439]: Remounting '/' read-only in with options 'lowerdir=run/initramfs/ro,upperdir=run/initramfs/rw/cow,workdir=run.
[  254.594300] [1440]: Remounting '/run/initramfs/rw' read-only in with options '(null)'.
[  254.606979] systemd-shutdown[1]: All filesystems unmounted.
[  254.612657] systemd-shutdown[1]: Deactivating swaps.
[  254.618250] systemd-shutdown[1]: Failed to parse (null): No such file or directory
[  254.626090] systemd-shutdown[1]: Failed to deactivate swaps: No such file or directory
[  254.634192] systemd-shutdown[1]: Detaching loop devices.
[  254.685542] watchdog: watchdog0: watchdog did not stop!
[  254.690824] shutdown: 6 output lines suppressed due to ratelimiting
shutdown: reboot --timeout 90000000us --log-level 6 --log-target kmsg --log-color
+ sort -r
+ awk '/oldroot|mnt/ { print $2 }'
+ umount /oldroot/sys/kernel/debug
+ umount /oldroot/sys/kernel/config
+ umount /oldroot/sys/fs/cgroup/unified
+ umount /oldroot/sys/fs/cgroup/systemd
+ umount /oldroot/sys/fs/cgroup
+ umount /oldroot/sys/fs/bpf
+ umount /oldroot/sys
+ umount /oldroot/proc
+ umount /oldroot/dev/shm
+ umount /oldroot/dev/pts
+ umount /oldroot/dev
+ umount /oldroot
+ umount /mnt/initramfs/rw
+ umount /mnt/initramfs/ro
+ umount /mnt
+ set +x
Pinging watchdog with args -t 1 -T 5
update: --clean-saved-files
[  255.271775] jffs2: notice: (1473) jffs2_build_xattr_subsystem: complete building xattr subsystem, 4 of xdatum (3 unchecked, 1 orp.
Updating bmc...
Erasing block: 512/512 (100%)

```

Thanks

mine260309, 2019-07-05:
This is a question, not an issue, so it is better to ask in OpenBMC mailing list, or in IRC.

And could you elaborate your question?
Are your asking where `/run/initramfs/shutdown` is called, or where the `update: --clean-saved-files` is called?

coffee2235, 2019-07-05:
I mean where /run/initramfs/shutdown is called.

I'm sorry about that .I will use mail or IRC to ask this question.

closed 2019-07-05 by coffee2235

mdmillerii, 2019-07-08:
To answer the question, https://www.freedesktop.org/wiki/Software/systemd/InitrdInterface/