openbmc/openbmc #3576: division by zero in occ driver

shenki, 2019-07-08:
Running 5.1.12-72dc922 on romulus and reading the occ hwmon status results in this:

```
[  795.372653] Division by zero in kernel.
[  795.376623] CPU: 0 PID: 2257 Comm: sensors Not tainted 5.1.12-72dc922 #1
[  795.383477] Hardware name: Generic DT based system
[  795.388285] Backtrace: 
[  795.390915] [<80107cf0>] (dump_backtrace) from [<80107f24>] (show_stack+0x20/0x24)
[  795.398509]  r7:00000000 r6:00000000 r5:00000000 r4:00000000
[  795.404335] [<80107f04>] (show_stack) from [<806839b0>] (dump_stack+0x20/0x28)
[  795.411720] [<80683990>] (dump_stack) from [<801080a8>] (__div0+0x20/0x28)
[  795.418706] [<80108088>] (__div0) from [<806821b8>] (Ldiv0+0x8/0x10)
[  795.425233] [<803581f8>] (div64_u64) from [<804f1c94>] (occ_get_powr_avg+0xa8/0xac)
[  795.433027]  r9:9585ab40 r8:959d4210 r7:959e2020 r6:9599ddb4 r5:00000000 r4:00000000
[  795.440922] [<804f1bec>] (occ_get_powr_avg) from [<804f2524>] (occ_show_power_a0+0x2e4/0x32c)
[  795.449575]  r5:9599ddb4 r4:9454ac30
[  795.453203] [<804f2240>] (occ_show_power_a0) from [<8041471c>] (dev_attr_show+0x2c/0x5c)
[  795.461425]  r10:00000001 r9:9585ab40 r8:959d4210 r7:9579d000 r6:8073df94 r5:9599ddb4
[  795.469331]  r4:9454ac30
[  795.472054] [<804146f0>] (dev_attr_show) from [<802aeaa8>] (sysfs_kf_seq_show+0x8c/0x104)
[  795.480393]  r5:00001000 r4:9454ac30
[  795.484022] [<802aea1c>] (sysfs_kf_seq_show) from [<802ad390>] (kernfs_seq_show+0x34/0x38)
[  795.492420]  r9:00000000 r8:9e381f60 r7:958500a0 r6:9454ac48 r5:9e381f60 r4:9454ac30
[  795.500376] [<802ad35c>] (kernfs_seq_show) from [<80260710>] (seq_read+0x15c/0x4dc)
[  795.508225] [<802605b4>] (seq_read) from [<802adeac>] (kernfs_fop_read+0x3c/0x1c0)
[  795.515963]  r10:00000000 r9:00000000 r8:9e381f60 r7:9e381f60 r6:802ade70 r5:00001000
[  795.523922]  r4:9585ab40
[  795.526527] [<802ade70>] (kernfs_fop_read) from [<8023ac4c>] (__vfs_read+0x4c/0x190)
[  795.534409]  r10:00000000 r9:00000000 r8:9e381f60 r7:9e381f60 r6:802ade70 r5:00001000
[  795.542386]  r4:958500a0
[  795.544965] [<8023ac00>] (__vfs_read) from [<8023ae30>] (vfs_read+0xa0/0x120)
[  795.552240]  r10:00000000 r9:00000000 r8:00000001 r7:9e381f60 r6:0034db18 r5:958500a0
[  795.560189]  r4:00001000
[  795.562765] [<8023ad90>] (vfs_read) from [<8023b24c>] (ksys_read+0x6c/0xf4)
[  795.569843]  r9:00000000 r8:00000000 r7:00001000 r6:0034db18 r5:958500a0 r4:958500a0
[  795.577621] [<8023b1e0>] (ksys_read) from [<8023b2ec>] (sys_read+0x18/0x1c)
[  795.584709]  r9:9e380000 r8:801011e4 r7:00000003 r6:437e11bc r5:437e1654 r4:0034c178
[  795.592610] [<8023b2d4>] (sys_read) from [<80101000>] (ret_fast_syscall+0x0/0x54)
[  795.600226] Exception stack(0x9e381fa8 to 0x9e381ff0)
[  795.605302] 1fa0:                   0034c178 437e1654 00000003 0034db18 00001000 00000000
[  795.613617] 1fc0: 0034c178 437e1654 437e11bc 00000003 000005e8 00000000 00000000 7eed5704
[  795.621913] 1fe0: 00000070 7eed50d0 4370c2b0 437649a0
[  796.086900] Division by zero in kernel.
```

The host firmware is open-power-romulus-v2.3-rc2-132-g8982825e, with occ-4b93968.

mine260309, 2019-07-08:
Reproduced on Romulus as well.
```
   1 static u64 occ_get_powr_avg(u64 *accum, u32 *samples)
   2 {
   3         return div64_u64(get_unaligned_be64(accum) * 1000000ULL,
   4                          get_unaligned_be32(samples));
   5 }
```
The code does not verify if `*samples` is zero or not.
Possibly, make it return 0 if `*samples` is 0?

mine260309, 2019-07-08:
A quick patch is sent: https://lists.ozlabs.org/pipermail/openbmc/2019-July/016975.html

closed 2020-02-06 by stale[bot]