openbmc/openbmc #3577: kernel git image size exceeds the limit

prakeerthi, 2019-07-08:


prakeerthi, 2019-07-08:
HI , I compiled the  image for my AST based board. I can observe the below errors:
 SQUASHFS error: Unable to read page, block 9b62d4, size ff2c
[   97.211282] SQUASHFS error: Unable to read data cache entry [9b62d4]
[   97.217649] SQUASHFS error: Unable to read page, block 9b62d4, size ff2c
[   97.224457] SQUASHFS error: Unable to read data cache entry [9b62d4]
[   97.230903] SQUASHFS error: Unable to read page, block 9b62d4, size ff2c

Th kernel fit image size also with in the limits. Need support on this issue

mdmillerii, 2019-07-09:
Your choices are to remove some content from the kernel fit (either a smaller initramfs or less kernel configuration) or to increase the size of the fit possibly by shrinking the space for the main squashfs (if you have space).

closed 2020-02-07 by stale[bot]