openbmc/openbmc #3578: Failed to create LDAP config due to file system error

anoo1, 2019-07-09:
@sivassrr reports the following error:

Step 1) Moved to master
Step 2) Log into openbmc GUI
Step 3) Selected following menu options: Access Control -> LDAP -> LDAP Page
Step 4) Disabled LDAP authentication - (earlier LDAP was enabled)
Step 5) Enable LDAP authentication and populate all the LDAP parameters including LDAP password
Step 6) Observed Save button enabled after all necessary parameters entered
Step 7) Tried to Save and hit with error "Unable to update LDAP settings".

Early journal messages:
```
[   13.824330] systemd[1]: Starting Create Static Device Nodes in /dev...
[   14.293756] systemd[1]: Started Create Static Device Nodes in /dev.
[   14.346890] systemd[1]: Starting udev Kernel Device Manager...
[   14.418690] systemd[1]: Reached target Local File Systems (Pre).
[   14.440113] systemd[814]: systemd-udevd.service: ProtectHostname=yes is configured, but the kernel does not support UTS namespaces, ignoring namespace setup.
[   15.436246] overlayfs: upper fs does not support xattr, falling back to index=off and metacopy=off
[   19.101393] systemd[1]: Started Journal Service.
[   19.438798] 8021q: adding VLAN 0 to HW filter on device eth0
[   19.444652] A link change request failed with some changes committed already. Interface eth0 may have been left with an inconsistent configuration, please check.
[   19.467339] ftgmac100 1e660000.ethernet eth0: NCSI: Handler for packet type 0x82 returned -19
[   28.840045] systemd-journald[825]: Received request to flush runtime journal from PID 1
[   28.850229] systemd-journald[825]: File /var/log/journal/3cfbe40a45c24e4091e87e226227ec16/system.journal corrupted or uncleanly shut down, renaming and replacing.
```

```
Jun 25 06:02:50 witherspoonSivas systemd[1]: Started Phosphor certificate manager for nslcd.
Jun 25 06:03:01 witherspoonSivas nslcd[1249]: nslcd: cannot open config file (/etc/nslcd.conf): Stale file handle
```

```
root@witherspoonSivas:~# ls -al /etc/
ls: /etc/nslcd.conf: Stale file handle
-rw-r--r--    1 root     root          4787 Jun 25 07:20 nslcd.conf.default

root@witherspoonSivas:~# ls -l /var/persist/etc/
-rw-r-----    1 root     root           574 Jun 12 23:48 nslcd.conf
```

anoo1, 2019-07-09:
Full journal attached. Seems the /etc/ overlay failed to unmount during reboot, this may lead to corruption.
@mdmillerii You had noticed the rwfs filesystem (/var) and overlay (/etc) were not in fstab, so they wouldn't be unmounted by systemd. Could this be part of the issue?

```
	"MESSAGE" : "Failed unmounting /var.",
	"MESSAGE" : "var.mount: Mount process exited, code=exited, status=32/n/a",
	"MESSAGE" : "umount: /var: target is busy."
	"MESSAGE" : "Failed unmounting /etc.",
	"MESSAGE" : "etc.mount: Mount process exited, code=exited, status=32/n/a",
```

[journal.log](https://github.com/openbmc/openbmc/files/3373511/journal.log)

mdmillerii, 2019-07-09:
I don't know if adding them to `mtab` will allow systemd to unmount them; it may not close references to `/etc`.  But we can try it.

sivassrr, 2019-07-16:
Tried to run LDAP related test on latest master:  and not able to read nslcd.conf due to following error. And LDAP config exists but LDAP login failed. ```root@witherspoon-YL30UF74T00J:~# cat /etc/nslcd.conf
[ 2095.447539] overlayfs: failed to get inode (-116)
[ 2095.452522] overlayfs: failed to get inode (-116)
cat: can't open '/etc/nslcd.conf': Stale file handle
root@witherspoon-YL30UF74T00J:~# cat /etc/os-release
ID="openbmc-phosphor"
NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro)"
VERSION="2.8.0-dev"
VERSION_ID="2.8.0-dev-60-gaf70bf502"
PRETTY_NAME="Phosphor OpenBMC (Phosphor OpenBMC Project Reference Distro) 2.8.0-dev"
BUILD_ID="2.8.0-dev"
OPENBMC_TARGET_MACHINE="witherspoon"
root@witherspoon-YL30UF74T00J:~#```

sivassrr, 2019-07-16:
Have sent FFDC to @anoo1

sivassrr, 2019-07-17:
With 2.7 master able to configure secure LDAP successfully. Able to login 
``` curl --insecure -X POST -D headers.txt https://${bmc}/redfish/v1/SessionService/Sessions -d '{"UserName":"sivasjxp", "Password":"india@123"}'
{
  "@odata.context": "/redfish/v1/$metadata#Session.Session",
  "@odata.id": "/redfish/v1/SessionService/Sessions/3gaL21mLPW",
  "@odata.type": "#Session.v1_0_2.Session",
  "Description": "Manager User Session",
  "Id": "3gaL21mLPW",
  "Name": "User Session",
  "UserName": "sivasjxp"
} ```
======================================
And able to power on / power off as well. Didnt try group / privileges as it is not implemented yet. 
```cat headers.txt
HTTP/1.1 201 Created
X-Auth-Token: gkixETMJMlrpiikiKC39
Location: /redfish/v1/SessionService/Sessions/3gaL21mLPW
Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
X-Frame-Options: DENY
Pragma: no-cache
Cache-Control: no-Store,no-Cache
Content-Security-Policy: default-src 'self'; img-src 'self' data:
X-XSS-Protection: 1; mode=block
X-Content-Type-Options: nosniff
X-UA-Compatible: IE=11
Content-Type: application/json
Server: iBMC
Date: Tue, 16 Jul 2019 15:15:47 GMT
Content-Length: 287

sivass-mbp:LDAPConfigurationActual sivas$ export bmc_token=gkixETMJMlrpiikiKC39
sivass-mbp:LDAPConfigurationActual sivas$ curl -k -H "X-Auth-Token: $bmc_token" -X GET https://${bmc}/redfish/v1/Systems/system/LogServices/EventLog/Entries
{
  "@odata.context": "/redfish/v1/$metadata#LogEntryCollection.LogEntryCollection",
  "@odata.id": "/redfish/v1/Systems/system/LogServices/EventLog/Entries",
  "@odata.type": "#LogEntryCollection.LogEntryCollection",
  "Description": "Collection of System Event Log Entries",
  "Members": [],
  "Members@odata.count": 0,
  "Name": "System Event Log Entries"
}sivass-mbp:LDAPConfigurationActual sivas$ curl -k -H "X-Auth-Token: $bmc_token"-X GET https://${bmc}/redfish/v1/Systems/system
{
  "@odata.context": "/redfish/v1/$metadata#ComputerSystem.ComputerSystem",
  "@odata.id": "/redfish/v1/Systems/system",
  "@odata.type": "#ComputerSystem.v1_6_0.ComputerSystem",
  "Actions": {
    "#ComputerSystem.Reset": {
      "ResetType@Redfish.AllowableValues": [
        "On",
        "ForceOff",
        "ForceOn",
        "ForceRestart",
        "GracefulRestart",
        "GracefulShutdown",
        "PowerCycle"
      ],
      "target": "/redfish/v1/Systems/system/Actions/ComputerSystem.Reset"
    }
  },
  "AssetTag": "",
  "BiosVersion": "IBM-witherspoon-OP9-v2.3-9.44",
  "Boot": {
    "BootSourceOverrideEnabled": "Disabled",
    "BootSourceOverrideMode": "Legacy",
    "BootSourceOverrideTarget": "None",
    "BootSourceOverrideTarget@Redfish.AllowableValues": [
      "None",
      "Pxe",
      "Hdd",
      "Cd",
[... 122 lines cut ...]
      "State": "Disabled"
    },
    "TotalSystemMemoryGiB": 0
  },
  "Model": "8286-42B",
  "Name": "system",
  "PartNumber": "",
  "PowerState": "On",
  "ProcessorSummary": {
    "Count": 0,
    "Status": {
      "State": "Disabled"
    }
  },
  "Processors": {
    "@odata.id": "/redfish/v1/Systems/system/Processors"
  },
  "SerialNumber": "13184DA",
  "Status": {
    "Health": "OK",
    "HealthRollup": "OK",
    "State": "Enabled"
  },
  "SystemType": "Physical"
}```

sivassrr, 2019-07-23:
One more time we hit with Filesystem error. But dont know the reason, One of my team mate was using and I tried to do codeupdate found it is failing. But file size size looks good to me. 
```root@wsbmc013:~# df
Filesystem           1K-blocks      Used Available Use% Mounted on
/dev/root                17280     17280         0 100% /
devtmpfs                181440         0    181440   0% /dev
ubi0:rwfs                 4644      1280      3100  29% /var
overlay                   4644      1280      3100  29% /etc
tmpfs                   214720         8    214712   0% /dev/shm
tmpfs                   214720       784    213936   0% /run
tmpfs                   214720         0    214720   0% /sys/fs/cgroup
tmpfs                   214720       192    214528   0% /tmp
tmpfs                   214720         0    214720   0% /var/volatile
tmpfs                   214720         0    214720   0% /media
ubi0:rwfs                 4644      1280      3100  29% /home
tmpfs                   214720         0    214720   0% /usr/local
overlay                 214720         0    214720   0% /srv
/dev/ubiblock4_0         17280     17280         0 100% /media/rofs-6558405b
ubi4:rwfs                 2808      1656      1152  59% /media/alt/var
/dev/ubiblock8_0         27008     27008         0 100% /media/pnor-ro-e6348a51
ubi8:pnor-rw-e6348a51
                         13816       348     12720   3% /media/pnor-rw-e6348a51
ubi8:pnor-prsv             972        72       832   8% /media/pnor-prsv
/dev/ubiblock8_4         26496     26496         0 100% /media/pnor-ro-f01ebecf
ubi8:pnor-rw-f01ebecf
                         13816       352     12716   3% /media/pnor-rw-f01ebecf
ubi8:pnor-patch          13816        16     13052   0% /usr/local/share/pnor
```

sivassrr, 2019-07-23:
[dmesg3578.docx](https://github.com/openbmc/openbmc/files/3420384/dmesg3578.docx)
@anoo1 Let know if you need any further info. Have collected FFDC and kept.

sivassrr, 2019-07-23:
BMC level ```2.7.0-dev-1040-g616087d```

sivassrr, 2019-07-23:
Are we getting new kernel @anoo1

closed 2020-02-21 by stale[bot]