openbmc/openbmc #3579: ipmi: Unable to establish IPMI v2 / RMCP+ session

shenki, 2019-07-10:
Using master (e4f02c2ea) on Romulus, I'm unable to talk IPMI:

```
$ ipmitool -I lanplus -C 3 -U root -P 0penBmc -H bmc lan print
Error: Unable to establish IPMI v2 / RMCP+ session
```

shenki, 2019-07-10:
```
bmc:~# systemctl status phosphor-ipmi-net@eth0.service | cat
* phosphor-ipmi-net@eth0.service - Network IPMI daemon
   Loaded: loaded (/lib/systemd/system/phosphor-ipmi-net@.service; enabled; vendor preset: disabled)
   Active: active (running) since Wed 2019-07-10 13:34:46 UTC; 6min ago
 Main PID: 1303 (netipmid)
   CGroup: /system.slice/system-phosphor\x2dipmi\x2dnet.slice/phosphor-ipmi-net@eth0.service
           `-1303 /usr/bin/netipmid -c eth0

Jul 10 13:34:46 ozrom2-bmc systemd[1]: Started Network IPMI daemon.
Jul 10 13:34:48 ozrom2-bmc netipmid[1303]: Failed to get bus name
```

shenki, 2019-07-10:
```
bmc:~# netstat -lunt
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       
tcp        0      0 0.0.0.0:5355            0.0.0.0:*               LISTEN      
tcp        0      0 127.0.0.1:5900          0.0.0.0:*               LISTEN      
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      
udp        0      0 0.0.0.0:35222           0.0.0.0:*                           
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           
udp        0      0 0.0.0.0:5355            0.0.0.0:*                           
udp        0      0 127.0.0.53:53           0.0.0.0:*                           
udp        0      0 10.61.161.192:68        0.0.0.0:*   
```
A machine running a much older firmware is listening on 623:
```
udp        0      0 :::623                  :::*                                
```

I am able to ssh to the machine, so network is working.

shenki, 2019-07-10:
I stopped the service and started netipmid on the command line. There is no further output:

```
bmc:~# /usr/bin/netipmid -c eth0 -v 1
```

from my machine I am unable to connect to the ipmi port with `ipmitool` or `nc`.

shenki, 2019-07-10:
A connection is being established. I ran netipmid under strace and saw traffic. I was also able to capture the dbus traffic, which is attached to this comment.
[dbus-traffic.txt](https://github.com/openbmc/openbmc/files/3377985/dbus-traffic.txt)

mine260309, 2019-07-11:
Most likely it's due to the permission issue.
An old BMC has `/etc/group` in rwfs, and it's preserved during code update.
After updating to a new BMC that enables user management, the preserved `/etc/group` cause the `root` user has unexpected, in this case, it's missing the `ipmi` group.

The new code should really fix this issue, otherwise, the end user will have to do factory reset to get it working.

shenki, 2019-07-11:
I confirmed that the root user was only a member of the root group. I removed the copy of /etc/groups from the rwfs (`rm /run/initramfs/rw/cow/etc/groups*`) and rebooted, and ipmitool worked.

Could the ipmi daemon have added a warning to log that access was denied due to permission failure? That would have made this easier to debug.

We should also look at fixing the issue of someone upgrading a BMC from non-group to group as Lei said.

closed 2020-02-09 by stale[bot]