openbmc/openbmc #3581: Build OpenBMC Raspberry PI with bmcweb?

alanmimms, 2019-07-19:
Has anyone built the meta-raspberrypi target for OpenBMC with some patch to build the `bmcweb` application into the resulting image? I've spent some time trying to learn enough about Yocto to do this myself, but it's a steep learning curve. I see evidence that Intel, IBM, and others have builds for their platforms with `bmcweb`. I think I just need a little help connecting the dots here. TIA.

geissonator, 2019-07-22:
Hi @alanmimms . In general questions like this should go on the mailing list or on IRC.

Some work was done in the past with rasberry pi. The changes have not had much activity though:
https://gerrit.openbmc-project.xyz/q/topic:%22raspberrypi%22+(status:open%20OR%20status:merged)

Adding bmcweb into a package should happen by default now with meta-phosphor layer being included.   See https://github.com/openbmc/meta-phosphor/blob/master/recipes-phosphor/packagegroups/packagegroup-obmc-apps.bb#L40

alanmimms, 2019-07-22:
Thanks. I'll ask on mailing list in future.

I still don't have my answer, as far as I can tell. Doing `bitbake core-image-full-cmdline` yields no bmcweb. I see a fix for correcting override prioritization of Phosphor at https://gerrit.openbmc-project.xyz/c/openbmc/meta-evb/+/17058 but it isn't included in the sources yet AFAICT. I may experiment with this.

But I'm far from confident that I would get a full Phosphor layer by using that `bitbake core-image-full-cmdline` command. I just don't have a clear view of how all of these pieces thread together into a build based on the bitbake target.

alanmimms, 2019-07-22:
Yes. I merged the changes from https://gerrit.openbmc-project.xyz/c/openbmc/meta-evb/+/17058 and appended the additional tweak to my `local.conf` to avoid the problem with the missing `phosphor-gpio-keys` file:

`KERNEL_FEATURES_remove = "phosphor-gpio-keys"
`

This built nicely, and I now have `bmcweb` doing Redfish things on my raspberrypi2 target as desired. I haven't tried the other Raspberry Pi targets yet, and I have done no exhaustive testing. But it does work for me at least for a smoke test.

Thank you for taking the time to help me get this going.

closed 2019-07-22 by alanmimms