openbmc/openbmc #3585: fail to build openBMC

wequal, 2019-07-27:
I am a flasher and trying to build TEMPLATECONF=meta-aspeed/conf/ without any changing, got the following errors, who can help me out, thanks.

| checking target system type... i686-pc-linux-gnu
| configure: error: internal configure error for the platform triplet, please file a bug report
| NOTE: The following config.log files may provide further information.
| NOTE: /home/wequal/obmc/openbmc/build/tmp/work/i686-linux/python3-native/3.7.3-r0/build/config.log
| ERROR: configure failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_configure (log file is located at /home/wequal/obmc/openbmc/build/tmp/work/i686-linux/python3-native/3.7.3-r0/temp/log.do_configure.3743)
ERROR: Task (virtual:native:/home/wequal/obmc/openbmc/meta/recipes-devtools/python/python3_3.7.3.bb:do_configure) failed with exit code '1'
NOTE: Tasks Summary: Attempted 435 tasks of which 390 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/wequal/obmc/openbmc/meta/recipes-devtools/python/python3_3.7.3.bb:do_configure

geissonator, 2019-07-29:
I've never tried building that target, could you try something like "export TEMPLATECONF=meta-evb/meta-evb-aspeed/meta-evb-ast2500/conf"? A quick search brought up http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=9d2ea1ea50355987fd4a0651817fd969cd838acc which looks somewhat similar. You're using the latest openbmc/openbmc from master branch? All of our CI testing is on Ubuntu 18, but I don't know of anything preventing Ubuntu 14 from working.

wequal, 2019-07-30:
After change the os from UBUNTU 14.04.6 desktop i386 to UBUNTU 14.04.06 desktop amd64, the issue is gone.
I also successfully compile the firmware on the ubuntu 18.04.2 desktop amd64.   
thanks Geissonator.

closed 2019-07-30 by wequal