openbmc/openbmc #3587: fail to run "runqemu evb-ast2500"

wequal, 2019-07-30:
I have successfully compiled the openBMC image, but fail to runqemu, I don't have have experience on yocto and runqemu, is there any document for the runquem configuration?

obmc@obmc-vpc:~/Desktop/obmc/openbmc/build$ bitbake core-image-minimal
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb should set LAYERSERIES_COMPAT_evb in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-aspeed should set LAYERSERIES_COMPAT_evb-aspeed in its conf/layer.conf file to list the core layer names it is compatible with.
WARNING: Layer evb-ast2500 should set LAYERSERIES_COMPAT_evb-ast2500 in its conf/layer.conf file to list the core layer names it is compatible with.
Loading cache: 100% |###################################################################################################################| Time: 0:00:00
Loaded 3443 entries from dependency cache.
Parsing recipes: 100% |#################################################################################################################| Time: 0:00:02
Parsing of 2348 .bb files complete (2347 cached, 1 parsed). 3444 targets, 339 skipped, 0 masked, 0 errors.
WARNING: No bb files matched BBFILE_PATTERN_evb ''
WARNING: No bb files matched BBFILE_PATTERN_evb-aspeed ''
WARNING: No bb files matched BBFILE_PATTERN_evb-ast2500 ''
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.43.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "arm-openbmc-linux-gnueabi"
MACHINE              = "evb-ast2500"
DISTRO               = "openbmc-phosphor"
DISTRO_VERSION       = "0.1.0"
TUNE_FEATURES        = "arm armv6 thumb arm1176jzs"
TARGET_FPU           = "soft"
meta                 
meta-poky            
meta-oe              
meta-networking      
meta-python          
meta-webserver       
meta-phosphor        
meta-aspeed          
meta-evb             
meta-evb-aspeed      
meta-evb-ast2500     = "master:04508dc9f4d6014e9a85dffd39065a433847b68b"

Initialising tasks: 100% |##############################################################################################################| Time: 0:00:01
Sstate summary: Wanted 3 Found 0 Missed 3 Current 555 (0% match, 99% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 1865 tasks of which 1853 didn't need to be rerun and all succeeded.

Summary: There were 9 WARNING messages shown.
obmc@obmc-vpc:~/Desktop/obmc/openbmc/build$ runqemu evb-ast2500
runqemu - INFO - Running MACHINE=evb-ast2500 bitbake -e...
runqemu - ERROR - /home/obmc/Desktop/obmc/openbmc/build/tmp/deploy/images/evb-ast2500 not a directory valid DEPLOY_DIR_IMAGE
ls: cannot access '/home/obmc/Desktop/obmc/openbmc/build/tmp/deploy/images/evb-ast2500/*.qemuboot.conf': No such file or directory
runqemu - ERROR - Command 'ls -t /home/obmc/Desktop/obmc/openbmc/build/tmp/deploy/images/evb-ast2500/*.qemuboot.conf' returned non-zero exit status 2.
runqemu - INFO - Cleaning up
obmc@obmc-vpc:~/Desktop/obmc/openbmc/build$

amboar, 2019-07-30:
Hi @wequal, unfortunately we don't have good integration of our ASPEED QEMU machines with yocto. If you're looking to contribute to the project this would be a handy place to start!

Separately, there's some [documention in the cheatsheet](https://github.com/openbmc/docs/blob/master/cheatsheet.md#using-qemu) on how to get an OpenBMC image up and running under QEMU, hopefully that helps you out.

Finally, in the future please make sure to take advantage of [github's markdown formatting](https://help.github.com/en/articles/basic-writing-and-formatting-syntax#quoting-code) when posting console output, this will cleanup the strike-throughs in your post above.

wequal, 2019-07-30:
thanks Amboar, I will study the document and close this issue.

closed 2019-07-30 by wequal