openbmc/openbmc #3590: How can I add python tools in image now?

shrd20190131, 2019-08-07:
After I add python3 in obmc-phosphor-image.bbappend, the error below will appear.
`OBMC_IMAGE_EXTRA_INSTALL_append += " ipmitool"
OBMC_IMAGE_EXTRA_INSTALL_append += " python3"`

`ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Image '/home/usird2/workspace/dean/openbmc-2.8.0-dev/build/tmp/work/jbof-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/deploy-obmc-phosphor-image-image-complete/obmc-phosphor-image-jbof.squashfs-xz' is too large!
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static:
ERROR: obmc-phosphor-image-1.0-r0 do_generate_static: Function failed: do_generate_static
ERROR: Logfile of failure stored in: /home/usird2/workspace/dean/openbmc-2.8.0-dev/build/tmp/work/jbof-openbmc-linux-gnueabi/obmc-phosphor-image/1.0-r0/temp/log.do_generate_static.19788
ERROR: Task (/home/usird2/workspace/dean/openbmc-2.8.0-dev/meta-phosphor/recipes-phosphor/images/obmc-phosphor-image.bb:do_generate_static) failed with exit code '1'
`

geissonator, 2019-08-07:
This is the reason we needed to make python optional. On most OpenBMC systems, the flash chip is not large enough to contain all of the OpenBMC features + python.  Your generated image does not fit within the flash chip size ("**obmc-phosphor-image-jbof.squashfs-xz' is too large!**"

You'll have to find something to remove to make space. https://openpower.xyz/job/openbmc-build/distro=ubuntu,label=builder,target=witherspoon/lastSuccessfulBuild/artifact/deploy/images/results.txt shows the sizes of each of the files. If you don't need a web server then removing bmcweb is one way to get a lot of extra space.

shrd20190131, 2019-08-08:
> This is the reason we needed to make python optional. On most OpenBMC systems, the flash chip is not large enough to contain all of the OpenBMC features + python. Your generated image does not fit [...]

Thank you for your answer!

closed 2019-08-08 by shrd20190131

kevinlee0505, 2019-11-11:
Hi,
I have same image problem when I tried to build OpenBMC to Raspberry Pi.
But, I modified manually *meta-phosphor/classes/image_types_phosphor.bbclass* to  extend flash size. It can fix that problem.
```
$ git diff
diff --git a/meta-phosphor/classes/image_types_phosphor.bbclass b/meta-phosphor/classes/image_types_phosphor.bbclass
index f7742c8..28035d1 100644
--- a/meta-phosphor/classes/image_types_phosphor.bbclass
+++ b/meta-phosphor/classes/image_types_phosphor.bbclass
@@ -28,7 +28,7 @@ IMAGE_TYPES_MASKED += "mtd-static mtd-static-alltar mtd-static-tar mtd-ubi mtd-u
 
 # Flash characteristics in KB unless otherwise noted
 DISTROOVERRIDES .= ":flash-${FLASH_SIZE}"
-FLASH_SIZE ?= "32768"
+FLASH_SIZE ?= "131072"
 FLASH_PEB_SIZE ?= "64"
 # Flash page and overhead sizes in bytes
 FLASH_PAGE_SIZE ?= "1"
```

229024527, 2020-08-20:
Hi can you tell me how to use qemu simulate this generated file

mdmillerii, 2020-08-20:
https://github.com/openbmc/docs/blob/master/development/README.md