openbmc/openbmc #3591: ipmitool print FRU data device not present(Unspecified error)--EVB2500ASPEED

Dan00121, 2019-08-08:
Hi all ,

I have installed openbmc on evb-ast2500 board and use devtool to add ipmitool into it.
The ipmitool was added from [https://github.com/openbmc/ipmitool](url)
I don't know why I can't print FRU data and it showed that  Device not present (Unspecified error)
And I was also unable to write FRU data from ipmitool fru write command.

**EEPROM Location**
```
root@evb-ast2500:/sys/devices/platform/ahb/ahb:apb/ahb:apb:i2c@1e78a000/1e78a100.i2c-bus/i2c-3/3-0050# ls
3-00500    driver     eeprom     modalias   name       of_node    subsystem
```
**ipmitool fru print** 
```
root@evb-ast2500:~# ipmitool fru print
FRU Device Description : Builtin FRU Device (ID 0)
FRU Device Description : dimm0 (ID 1)
 Device not present (Unspecified error)

FRU Device Description : dimm1 (ID 2)
 Device not present (Unspecified error)

FRU Device Description : cpu0 (ID 3)
 Device not present (Unspecified error)

FRU Device Description : cpu1 (ID 4)
 Device not present (Unspecified error)
```
**ipmitool fru print 0**
```
root@evb-ast2500:~# ipmitool fru print 0
 Device not present (Unspecified error)
```
**ipmitool fru write 0 /tmp/w.bin**  This command shows nothing.  
```
root@evb-ast2500:/tmp# ipmitool fru write 0 /tmp/w.bin
root@evb-ast2500:/tmp# ipmitool fru print 0
 Device not present (Unspecified error)
```

Thanks for help.

geissonator, 2019-08-08:
I'm not super familiar with the evb-ast2500 but a lot of the FRU device information comes from the BIOS on most servers. So on a real server, you boot the system once to get the BIOS to send down all the FRU information to the BMC. The BMC then caches that information and uses it for ipmitool fru commands. From a write aspect, I've never tried that. Is it something that should work in OpenBMC @tomjoseph83 ?

tajudheenk, 2019-09-04:
If your are trying to read from an eeprom based FRU, make sure that you have proper `SYSFS_PATH` mapped to appropriate eeprom.

Refer following files for similar implementation details.
[meta-quanta/meta-f0b/recipes-phosphor/ipmi/phosphor-ipmi-fru_%.bbappend](https://github.com/openbmc/openbmc/blob/master/meta-quanta/meta-f0b/recipes-phosphor/ipmi/phosphor-ipmi-fru_%25.bbappend)
[meta-quanta/meta-f0b/recipes-phosphor/ipmi/phosphor-ipmi-fru/obmc/eeproms/system/motherboard](https://github.com/openbmc/openbmc/blob/master/meta-quanta/meta-f0b/recipes-phosphor/ipmi/phosphor-ipmi-fru/obmc/eeproms/system/motherboard)

Also make sure that you have ipmi compatible FRU binary flashed on eeprom. 

To view the content of FRU eeprom you may try 
`hexdump -C /sys/bus/i2c/devices/3-0050/eeprom`

For FRU write you may try (Note: This command will corrupt your FRU if you do not use a correct file)
`dd  if=<FRU binary file> of=/sys/bus/i2c/devices/3-0050/eeprom`

closed 2019-09-11 by Dan00121

229024527, 2020-09-25:
Hi can you tell me how to generate a w.bin file

tajudheenk, 2020-09-30:
> Hi can you tell me how to generate a w.bin file

If you are asking about ipmi fru binary file generation, please refer to 
[https://github.com/ipmitool/frugen](https://github.com/ipmitool/frugen)

wu790607, 2021-04-27:
Hi Dan00121 

Can you provide your solution?
I can't write FRU data via ipmitool fru write command either.

Thanks!!